pybricks/support #1278: [Bug] Observing crashes the City Hub

laurensvalk, 2023-11-06:
**Describe the bug**
After just a few minutes, the City Hub resets itself (consistently) on receiving data:

Send with Prime Hub:
```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import ColorLightMatrix
from pybricks.parameters import Color, Port
from pybricks.tools import wait

hub = PrimeHub(broadcast_channel=12)

EXTRA_DATA = "extra bytes to send"

while True:
    for i in range(100):
        # hub.ble.broadcast(i)
        if i % 2:
            hub.ble.broadcast(("red", EXTRA_DATA))
            hub.light.on(Color.RED)
        else:
            hub.ble.broadcast(("green", EXTRA_DATA))
            hub.light.on(Color.GREEN)
        wait(250)
```

Receive on City Hub:

```python
from pybricks.hubs import CityHub
from pybricks.parameters import Color
from pybricks.tools import wait

hub = CityHub(observe_channels=[12])

while True:
    data = hub.ble.observe(12)
    print(data)
    if data is None:
        # bad state
        hub.light.on(Color.VIOLET)
    else:
        if data[0] == "red":
            hub.light.on(Color.RED)
        else:
            hub.light.on(Color.GREEN)
    wait(10)
```

laurensvalk, 2023-11-06:
Reducing the amount of data seems to delay or prevent the crash:

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import ColorLightMatrix
from pybricks.parameters import Color, Port
from pybricks.tools import wait

hub = PrimeHub(broadcast_channel=12)

while True:
    for i in range(100):
        hub.ble.broadcast("red" if i % 2 else "green")
        hub.light.on(Color.RED if i % 2 else Color.GREEN)
        wait(250)
```

```python
from pybricks.hubs import CityHub
from pybricks.parameters import Color
from pybricks.tools import wait

hub = CityHub(observe_channels=[12])

while True:
    data = hub.ble.observe(12)
    if data is None:
        # bad state
        hub.light.on(Color.VIOLET)
    else:
        if data == "red":
            hub.light.on(Color.RED)
        else:
            hub.light.on(Color.GREEN)
    wait(10)
```

dlech, 2023-11-12:
[issue-1276-crash.zip](https://github.com/pybricks/support/files/13329284/issue-1276-crash.zip)

Message capture shows that the hub is sending `GAP_DEVICE_DISCOVERY_CANCEL` which means the MicroPython program has crashed and called `pb_type_BLE_cleanup()`. So, I don't think the hub is actually crashing, only the MicroPython program. So maybe a memory fragmentation problem?

dlech, 2023-11-12:
If we make the first string the same size (i.e. change `"green"` to `"grn"` so it is the same size as `"red"`) it appears to avoid the crash as well. This leads me to believe that the data buffer is somehow getting updated while we are iterating on it somewhere or something like that which causes an exception to be raised due to bad data.

closed 2023-11-12 by dlech

laurensvalk, 2023-11-13:
Awesome, thanks!