pybricks/support #1280: Pycharm Support

shmezi, 2023-11-09:
I feel like this is such a great tool, however the editor is nowhere near the capability of an actual IDE. I'd love to see a Pycharm extension made for this. I would love to contribute if something like this is started.

dlech, 2023-11-09:
The Pybricks core developers don't use Pycharam. But if anyone from the community wants to create an extension, we are happy to answer questions about the communication protocol.

shmezi, 2023-11-10:
I haven't really tried making extensions but I'd be happy to try and make one, I just could never find where and how you communicate with the brick.

dlech, 2023-11-10:
> I just could never find where and how you communicate with the brick.

https://github.com/pybricks/technical-info/blob/master/pybricks-ble-profile.md is a good place to start

roykrikke, 2024-01-19:
I would love to see pycharm support also.