pybricks/support #1281: How to use remote light during an async function

Jedve, 2023-11-11:
**Question**
How can I change remote light color in a multitask ?

**Context**
When I try to change the Color of the remote connected to a CityHub, it works for the first call and crash on the second one with an error returned:
"OSError: This ressource cannot be used in two tasks at once."
When I do the same with the hub light, it works perfectly.

**Screenshots**
I put here a testing exemple that throw the error:
`hub = CityHub()

# Connect to the remote.
print('searching for remote...')
remote = Remote(timeout=None)
print(f'remote {remote.name()} found!')

async def change_color(color):
    remote.light.on(color=color)
    

async def main():
    for color in [Color.GREEN, Color.MAGENTA]:
        await change_color(color)
        await wait(1000)

# Runs the main program from start to finish.
run_task(main())`

laurensvalk, 2023-11-11:
The await keyword is missing in front of the on() call within your function.

This has since been added to the docs, but the updated docs haven't made it into a release yet.

Jedve, 2023-11-12:
@laurensvalk Thank you for your answer.

I tried it and, as a result, I don't get the error but the light color is not updated. Only the Color.GREEN is applied. If I replace the remote by the hub, it works only without the await before calling the on().

```python
async def change_color(color):
    print(f'Change color to {color}')
    await remote.light.on(color=color)
    print(f'Change color to {color} done')
    

async def main():
    for color in [Color.GREEN, Color.MAGENTA]:
        await change_color(color)
        await wait(1000)
```

The result of the added print() let me know that the function is called correctly:
[image]

But the light is still green.

laurensvalk, 2023-11-12:
Can you try it using the nightly firmware build? We recently fixed something that could impact this. Thanks!

https://nightly.link/pybricks/pybricks-micropython/workflows/build/master

To install, begin the firmware update as usual, but instead of selecting a hub on the first dialog, select "advanced", and pick the file you downloaded from the link above.

Jedve, 2023-11-12:
@laurensvalk 
It's working well now with this update. 
Thank you very much for your help.

So now, to use the remote light properly, do I wait for the next official release ?

laurensvalk, 2023-11-12:
You can keep using the version you just installed if you like. You'll be notified automatically when a new official version becomes available, including this fix.

Glad it worked!

closed 2023-11-12 by laurensvalk

mahil, 2023-12-28:
I also did install the latest nightly firmware build for the technic hub. But I always have problems when I use remote and async on the technic hub together.
This is just a simple program. The remote is connected just for a short time, and then it is disconnected without any error message.

[image]

The same program is running on a move hub without any issues.

laurensvalk, 2023-12-28:
Thanks @mahil - I've moved your note to a separate issue, which sounds a bit different from the documentation issue which was resolved here.