pybricks/support #1284: Getting Huskylens working on SPIKE Prime with Pybricks

weichunmei, 2023-11-12:
Hi All,

"I followed the steps in the post '[How to use the Huskylens with LEGO SPIKE Prime or Robot Inventor'](https://www.antonsmindstorms.com/2021/10/17/how-to-use-the-huskylens-with-lego-spike-prime-or-robot-inventor/), to use Huskylens on Spike Prime V3.0. When I copied the script from 'Library/SPIKEInstaller.py' and ran it on LEGO Education Spike V3.3.1 software, it prompted 'no module named 'machine''. I later found that it seems to be missing the 'Pybricks' library. How can I use non-official LEGO sensors on the Spike Prime V3.0 device?"@laurensvalk  @antonvh

antonvh, 2023-11-12:
Hi, this is not a Pybricks issue, so it doesn't really belong here. The problem is that neither Pybricks firmware nor LEGO SPIKE3 support a protocol that Huskylens uses: you need UART or I2C.  The best solution is to add an LMS-ESP32 board inbetween the Huskylens and the LEGO. That board is able to do voltage conversion, so the huskylens gets power. And the board is also abe to emulate the LEGO protocol, so the Huskylens can pose as a real LEGO sensor. https://www.antonsmindstorms.com/product/wifi-python-esp32-board-for-mindstorms/

Alternatively, you can downgrade to SPIKE2 or Inventor firmware.

datastormg, 2024-09-27:
I purchased this item from this store https://www.antonsmindstorms.com/product/wifi-python-esp32-board-for-mindstorms/ and 
 more than 20 days ago, and I still haven't received it.

laurensvalk, 2024-09-30:
I would recommend contacting their customer service. They're really nice folks, so I'm sure they're happy to help you with their products.