pybricks/support #1286: Troubleshooting Bluetooth in Pybricks Code: Connection but no Commands recognised

jmb-4, 2023-11-14:
Hi does someone know how to fix this:

I have a problem with the project shown on: https://pybricks.com/projects/tutorials/wireless/hub-to-device/pc-communication/ .

I’m working on a Raspberry Pi 4B 8Gb with Pycharm. I have the City Hub

I just copied the code online and pasted it in to the IDE. (and of course installed bleak)
And I saved the code for the hub on the hub using beta.pybricks.com

My Pi connects to the Hub, and I can start the programm on the Hub(button), but there is no action. .poll(0) isn‘t recognising anything, and the Pi doesn‘t receive anthing as well. (I tryed sending OK in the waiting loop on the Hub.)

Thanks to the Pybricks devellopers for the ability to program Lego Hubs!

laurensvalk, 2023-11-14:
I seem to remember we've changed this protocol some time since the last release and I wonder if this example still needs updating. :thinking: 

@dlech probably remembers this better than me. :smile:

jmb-4, 2023-11-14:
I had another hub laying around with an old pybricks version, so it works for me now.
But would be nice if the new pybricks version would support it too.

dlech, 2023-11-14:
Updated examples:
- https://github.com/orgs/pybricks/discussions/1108#discussioncomment-6153634
- https://github.com/orgs/pybricks/discussions/1108#discussioncomment-6167394