pybricks/support #1303: [Question] Controlling GeekServos with PU hubs?

gyenesvi, 2023-11-26:
I was wondering if PU hubs would be capable of controlling GeekServos.
GeekServos are regular RC servos in a Lego compatible housing, and they can be bought from many sources.

https://thepihut.com/products/geekservo-lego-compatible-180-degree-rotation-block-servo

They are cheap, small, powerful for the size, fast and precise, they come in 180 degree and continuous rotation versions. Sounds pretty surprising, but true, I have tried some, and many people build Lego RC vehicles with them. The only problem is they are hard to connect to existing Lego electronics systems, because of the connector they use, but most importantly the signal they need. So mostly they are used with custom RC electronics like RC transmitters/receivers, speed controllers and batteries.

GeekServos (like regular RC servos) work with PWM signals at 5-6V. For example, for the 180 degree variant, the width of the PWM signal determines their position, and they use 500-2500 usec range with some fixed frequency (say 50 Hz), 1500 usec meaning the center position, 500 usec being one end, and 2500 usec being the other end position. They are often programmed with things like Arduino or Raspberry Pi.

So I was wondering if the electronics in the PU hubs can be programmed to emit such PWM signals on one of the wires. If yes, I was thinking that the GeekServo could be rewired with 3rd party PU cables available online to be able to connect it to PU hubs. The GeekServo uses 3 wires, two for ground/power and one for the PWM control signal, so the 6 wires in the PU cable could do, probably two of them are used for ground/power anyway by the PU system, and one other could be used to send the PWM signal, the other 3 remaining unused.

Here are some programming examples:

https://docs.onion.io/omega2-maker-kit/maker-kit-servo-controlling-servo.html

Is this something the PU hubs would be capable of doing? Basically, it seems to me that given a fixed (50 Hz) frequency PWM signal, only the duty cycle percentage needs to be set appropriately (between 2,5% and 12,5%), which sounds like the same thing as using PU motors in basic Power mode. Is the PWM frequency of the Power mode known?

dlech, 2023-11-26:
If you are using EV3, then I would suggest something like this: http://www.mindsensors.com/ev3-and-nxt/25-8-channel-servo-controller-for-nxt-or-ev3

I'm not aware of anything like this for Powered Up though.

There isn't really a way to safely directly connect these kinds of servo motors to LEGO ports since the servos are 5V while the LEGO ports supply direct battery voltage. Also, on LEGO ports the PWM is on the power lines while servo motors need the PWM to be on a separate signal line. The firmware could be hacked to do this on pin 5 or 6 and it would be 3.3V instead of 5V.

So, yeah, it is really going to need some kind of electronics in between.

gyenesvi, 2023-11-26:
@dlech thanks for the response! Not using NXT, but Powered Up hub. And yeah, I was thinking of setting the value of pins 5/6 by the FW, and was hoping that the control circuitry would be running on 5V..

Interesting what can be done with the NXT though. Does that servo controller you referenced work because the NXT is using 5V for control signals?

One thing I don't quite understand is whether for servos the 5-6V range is mainly for the power line or is it needed for the control signal as well? Could they operate at lower voltage as well? Also, I've read that for the power line, at least for DC motors, using 6V is somewhat similar to using a higher voltage at a lower PWM duty cycle (say 9V at 60% duty cycle). So maybe using the battery voltage with 60% duty cycle and pin 5/6 for the PWM control signal at 3.3V could work out? What do you think?

dlech, 2023-11-26:
> Interesting what can be done with the NXT though. Does that servo controller you referenced work because the NXT is using 5V for control signals?

NXT/EV3 still requires the 3rd party hardware that I linked. 

> Could they [control signal] operate at lower voltage as well?

The specs you linked say: "Working Voltage: 3.3-6.0V". And in general, logic input don't need to be anywhere near full voltage (typically datasheets will say that 1.8V or 2V is enough for logic high of a "5V" input)

> So maybe using the battery voltage with 60% duty cycle and pin 5/6 for the PWM control signal at 3.3V could work out?

It depends on why the servo motor is rated at 4.8V. If it is because of the insulation on the wires in the motor, then using pulsed power at nearly twice the rated voltage probably isn't good for it. And there are also electronics in the servo that need to be continuously powered and not fried due to surpassing the rated voltage. So I wouldn't try it without some kind of voltage regulator.

gyenesvi, 2023-11-26:
> NXT/EV3 still requires the 3rd party hardware that I linked.

Yes I understand that, but I guessed that that board does not have a voltage regulator on it.

> The specs you linked say: "Working Voltage: 3.3-6.0V". And in general, logic input don't need to be anywhere near full voltage

Oh indeed, I was looking at another spec page that said 4.8-6V.. And also said 2kg.cm max torque.. So it varies depending on where I check it. Anyways, it seems it could work with 3.3V as well then. But I guess using one of pin 5/6 for constant 3.3V power and the other for the PWM signal would not work as the ground would not pair with the constant 3.3V right?

Anyways, I was thinking that if I use the PU hub with rechargeable batteries that is only 7.2V, not so far from the 6V, so that would hopefully not fry the servo (especially at 80% duty cycle). And I would even risk damaging one (it's cheap) just to try.

Would it be complicated to add FW support for controlling pin 5/6 with a PWM signal that can be parameterized as I want (while allowing to set duty cycle on the regular power pin at the same time)? I guess it could have a quite simple Python interface. Then it would be possible to experiment with or without voltage regulator.

dlech, 2023-11-26:
> Yes I understand that, but I guessed that that board does not have a voltage regulator on it.

It does - and it requires an external power supply for this.

> Would it be complicated to add FW support for controlling pin 5/6 with a PWM signal that can be parameterized as I want (while allowing to set duty cycle on the regular power pin at the same time)?

Yes. These pins are setup to be used as UART and GPIOs. But since it requires sub-millisecond precision, bit-banging the the GPIO isn't an option. So you would need to research the datasheet to see if any of the pins on any of the ports can be muxed to a TIMx PWM output. Then modify the firmware to disable regular use of that port and add your own driver for it. If none of the pins have the right mux, timer interrupts could be used, but it would be less accurate. Then add a new MicroPython class to provide an interface to the new driver from user code.

gyenesvi, 2023-11-26:
That does indeed sound more complicated than it would worth it. Anyways, thanks for all the info!

closed 2023-11-26 by gyenesvi