pybricks/support #1324: please post building instructions for train switch

jedierikb, 2023-12-10:
[image]

https://www.facebook.com/Pybricks/posts/pfbid02bswyh9dJQ7z9yFra1RAEqmFr6fcASWfRAYmNLcxR9FkaKnKLdFKRv5xpDFMDmuUMl?comment_id=349036371105744&reply_comment_id=865786858360958

laurensvalk, 2023-12-11:
Thanks for posting here! Much easier for us to keep track instead of Facebook :smile: 

We'll try to cover this in our next demo.

laurensvalk, 2023-12-11:
cc @KWSmit

laurensvalk, 2024-01-16:
Since I haven't gotten around to posting it yet, let me at least share the magic trick :smile: 

When you remove the yellow switch handle, you're left with the grey connector that has to be pushed left and right.

I've found that a round technic hole (beam, connector, etc) fits very well around it. From there, you have a lot of possibilities.

> Just to double check, this will be posted on YouTube?

Yes, but I'll post at least a link here.

KWSmit, 2024-01-20:
This is my best solution so far. Works fine with Medium Angular Motor and Technic Large Motor. I found that the Medium Linear Motor is not strong enough (so you have to use gears to give it more torque).
[image]
[image]