pybricks/support #1328: Generated Python code for Block `drivebase stop brake` should be `drivebase.brake()`, but is `drivebase.stop()`

laurensvalk, 2023-12-11:
### Discussed in https://github.com/orgs/pybricks/discussions/1326

<div type='discussions-op-text'>

<sup>Originally posted by **ggramlich** December 10, 2023</sup>
Currently the following code is generated:
[image]
[image]
[image]

Also the documentation for the `hold` option is missing - possibly because the Python translation is not straightforward (pun originally not intended).
[image]
</div>

laurensvalk, 2023-12-11:
Thanks for reporting. We recently [enabled brake for the drivebase](https://github.com/pybricks/support/issues/881) specifically for this reason, but it hasn't made it into the blocks yet.

For hold, indeed, there is technically not an equivalent, so we made it just drive 0 distance and hold there, which has the same effect.

ggramlich, 2024-02-18:
@laurensvalk 
Any chance to have the fix for the translation of the option "brake" to `drivebase.brake()` in one of the next updates?
It is not a big deal, I am just a bit worried that the kids might use the option "brake" and then with some update shortly before the competition the behavior of the robot changes slightly.
(I plan to switch to the stable version middle of March anyways.)

laurensvalk, 2024-02-18:
Thank you. Yes. With the big update for Xbox control out now, I'd like to start fixing as many small issues like these as possible.

laurensvalk, 2024-03-07:
Fixed and merged. Should be coming out in next release.

laurensvalk, 2024-03-07:
Thanks for reporting!

closed 2024-03-07 by laurensvalk