pybricks/support #1345: [Feature] Make Pybricks Code multilingual

laurensvalk, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
Pybricks can sometimes be harder to use for non-native English speakers.

**Describe the solution you'd like**
Most of the localization framework for this is already in place. Auto-translating the doc strings should now be easier than ever, with users perhaps able to review the results.

We'll also need to add a UI for this in the settings.

The documentation might be a bit trickier though.

Robo-Hinataya, 2024-02-08:
I teach robotics to children ages 7 to 15 in Japan.
I want to teach using block coding, but English blocks are difficult for children to understand.

Are there any plans to make block coding multilingual?
Or, I would be happy if I could freely set the characters of the block.

laurensvalk, 2024-02-08:
Yes, we have plans for this as part of a future update.

Thank you very much for offering to help! We will post an update in this thread with further information when we are closer to making this possible.

laurensvalk, 2024-02-08:
So far we have had translation requests for:

- Japanese
- Spanish
- Italian
- German
- French

As a first step, I think we can prepare the technical changes for this, and add one test translation to ensure we've covered all portions of the app and blocks. As a second step, we could organize a way to do the actual translations. Perhaps partially automated with human review.

Blockly has support for right-to-left languages too, but we'll need to review our Pybricks additions to ensure "block mutations/extensions" respect this as well.

biasedlogic, 2024-02-18:
Hi, I can support for German and Polish - especially interested in making the blocks multilingual.

wv77, 2024-05-27:
I would like to request support for Dutch as well. My children are now using Scratch (multilingual) and would love to use PyBricks.

juice-ontour, 2024-10-24:
Adding other languages would be a big step towards adaption of Pybricks by kids.
I can provide help, translation work, and review for German and French.

SamohtHub, 2024-12-28:
Hi, I am also interested in a German translation. The translations will allow children without knowledge of English to program in blocks. Similar to the Lego app. I am also willing to offer help for German. Is there already a status or plan until when the translations should be made possible?

wimvhex, 2025-01-02:
Hi, I'm interested in a Dutch translation. Feel free to contact me for helping in providing Dutch translations.

dlech, 2025-01-26:
German: https://github.com/pybricks/pybricks-code/pull/2331

Although, we'll probably want to use a translation service like [Crowdin](https://crowdin.com/) or [Weblate](https://weblate.org/en/) to help manage translations.

roykrikke, 2025-02-05:
+1 for Dutch translations support on my side.

sudefou, 2025-09-29:
N'hésitez pas à faire appel à moi pour la traduction en français. Merci. Philippe

suske-lambik, 2026-11-01:
Also very interested in this feature. 
I vote for support for Dutch language. 
I am willing to help with translation as well.

Or potentially with development, I have been in software development for 10+ years.

I would love this feature to teach my children and support them in using pybricks.

mjstassen, 2026-11-10:
Hi, what is the current status?
I would like to have Dutch language support and I am also interested in possibly helping with development.

laurensvalk, 2026-11-11:
Made some good progress, and most of the other prioritized app features are now complete.

We'd like to include a concept for translations in the upcoming beta version in a few weeks. We'll start with Dutch since that's my native language 😄.

When this is working, we'll prepare drafts in the languages discussed above for review. Thanks for your patience!