pybricks/support #1349: [Feature] Add ability to add comments

laurensvalk, 2023-12-22:
> Also, would be nice if there was a way to add comments on the block coding side.

_Originally posted by @southlakefll in https://github.com/pybricks/support/discussions/1348_

laurensvalk, 2023-12-22:
Thanks for this suggestion! This has been on our todo list but hasn't made it into a release yet.

There are at least three ways to do it:

1. Floating text boxes. This is the LEGO way.
2. Add a special block in which you can type (multiple lines of) text.
3. Add an expandable bubble on the block. This is the Blockly way.

---

Personal opinions:

1. I'm not the biggest fan of this one because these notes would get lost in the generated code. Because they're disconnected, they also tend to get in the way. Or in the wrong place as the program grows and you forget to move the comments along.

2. I think I like this option as it solves most problems of 1. But maybe this would get too big/messy? Feel free to comment your thoughts everyone.

3. The downside of this one is that you can't see it in screenshots. And because they're mostly hidden, it's easy to forget to update them so they may quickly get outdated.

laurensvalk, 2024-01-13:
Also requested by:

> Missing the possibility of adding comments on the block code side.

_Originally posted by @BertLindeman in https://github.com/pybricks/support/discussions/1377_

laurensvalk, 2024-02-08:
Also requested by:

> Moin zusammen, gibt es eine Möglichkeit, eigene Blockprogramme zu dokumentieren, ähnlich wie das unter Python geht. Text neben oder über den Befehlen.

_Originally posted by @msw-home in https://github.com/pybricks/support/discussions/1428_

ggramlich, 2024-02-11:
@laurensvalk 
I'd like to see your second option.
> Add a special block in which you can type (multiple lines of) text.

laurensvalk, 2024-02-11:
Here is a working concept for consideration in a future release.

I'm not yet sure what the best place for this block will be on the toolbox. We might add a new `utility` category for things like hub shutdown. Maybe there. Suggestions welcome!

[image]

BertLindeman, 2024-02-12:
I like the block like that, better than floating blobs of text.
But on the form of the connector I assume it will not connect to the setup.
I might want to add a comment there too, e.g. to explain why this motor is running counterclock.

Sorcerer13, 2024-03-16:
Has this functionality been implemented yet? I'm new to Pybricks Block Code, and being able to comment 'sample' code would help me to refresh my knowledge after a period away.

laurensvalk, 2024-03-28:
Everyone, this functionality is now ready to try out in https://beta.pybricks.com/

What do you think? Should this be included in the final version?

laurensvalk, 2024-03-28:
Here's how it looks:

[image]

laurensvalk, 2024-03-28:
@MonongahelaCryptidCooperative I noticed in your screenshots, the font color look different. Is that the case for all your devices?

BertLindeman, 2024-03-28:
> Everyone, this functionality is now ready to try out in https://beta.pybricks.com/
> 
> What do you think? Should this be included in the final version?

For sure.
We can always nag later about layout and fonts / bold or not. 😉 

The sooner the comment block is used the better.
Very good that the comments are not floating.

msw-home, 2024-03-28:
Hallo,

 

für mich wäre das ausreichend, um etwas zu dokumentieren.

Es wäre schön, wenn der eingegebene Text automatisch mit übersetzt werden.

Generell fehlt mir in der Hilfe, eine ausführlichere Beschreibung der einzelnen Blockbefehle. 

Ebenso würde ich ein paar mehr einfache Programme wie z.B., wie programmiert man eine Blinklichtanlage (Blinklicht Polizei) und so weiter, einstellen, in Blockprogrammierung.

Und immer dran denken, es sollen auch jüngere Kinder ab 8 Jahren inspiriert werden.  

 

LG Manfred

 

 

Mehr Funktionshinweise für die Blockbefehle, wären nützlich

 

 

Tel:    0151 26091955

Mail:   ***@***.***> ***@***.***

Sorcerer13, 2024-03-28:
Immediately (or at the latest by yesterday - whichever is the soonest!)

Seriously, please include it ASAP - the ability to insert comments is one
of the most important functions in any written coding language, as it
enables the programmer to express the intention of the code, be it code
snippets or the whole programme.
It's not an add-on - it's VITAL!

laurensvalk, 2024-03-28:
Thanks! You can already try it in action and document your code at https://beta.pybricks.com/

We always try out new features there first, and merge them into the stable version at https://code.pybricks.com/ when all the enthusiasts participating in discussions like these:
- are happy :smile: 
- don't experience any unexpected side effects :boom:

So far, so good!