pybricks/support #1350: [Feature] Add screenshot download option

laurensvalk, 2023-12-22:
> It would be nice if there was a built in way to export a clean image of the block coding (like a .png without the grid). There is a way to do this in the LEGO software (but it outputs an svg file which is not super easy to work with, and doesn't output the comments very well).

_Originally posted by @southlakefll in https://github.com/pybricks/support/discussions/1348_

laurensvalk, 2023-12-22:
Great idea. This was also requested by @DanieleBenedettelli.

---------

The good news is that our internal test version has this functionality already:

[image]

We'd still need to:
- Create a good UI in the main app.
- Add some padding. The current version above crops the result too much.

--------

For bonus points, we could encode the program **in** the download, so all you'd ever have to share is the picture and you could just open that. But maybe this is more confusing than helpful because images often get compressed when you send them by email or messaging apps.

southlakefll, 2023-12-22:
Sorry, I did a search but didn't see the post by DanieleBenedettelli. It's great that you are taking in feedback from users and adding features! I think it's already better than what LEGO offers. Thanks!

laurensvalk, 2023-12-22:
No worries, that request came by email. I just tagged him so he can follow along with our progress :)

laurensvalk, 2024-01-13:
Also requested by @BlueBillCanada in https://github.com/pybricks/support/discussions/1378_

Also with a good additional idea:

> Also, a white background with the dots removed would be nice.

laurensvalk, 2024-01-13:
This is getting to the top of the to-do list. Thanks everyone - this is definitely a great way to figure out what's most important. Keep the requests coming :)

DanieleBenedettelli, 2024-01-13:
The top would be having the program exported as PNG with transparent
background.

D

laurensvalk, 2024-01-18:
How's this? Coming up in the next release :smile: 

[image]

As a first step, I think we'll add just a very basic UI for this: a image-icon next to the other block UI buttons.

It might be nice to integrate it with the existing "download file" option in the future so that your PNG would automatically get the name of your program, but just having a download button is a good start.

ggramlich, 2024-01-18:
This looks fine to me. Also the transparent background allows to embed it on pages with dark mode.

I was thinking again about the idea of embedding the program into the PNG.
If you decide to do so, please make it optional, so that you can still download the python source and e.g. put it into version control.

laurensvalk, 2024-01-18:
We might have to contribute a fix to upstream Blockly to correctly show disabled blocks. https://github.com/google/blockly-samples/issues/2157

laurensvalk, 2024-01-18:
Everyone: you can try it with the latest beta over at: https://beta.pybricks.com/

The interface may change before the final release, but everyone can now easily make screenshots.

Thanks everyone for this suggestion!

[image]

BertLindeman, 2024-01-18:
Works, Laurens.

The size on screen in pybricks does not influence the size of the downloaded image. (No remark, just an observation)

laurensvalk, 2024-01-18:
That's right - that way you always get your screenshots in the same resolution, same margin, and no need to worry about forgetting to keep a block in view.

Some of the people in this thread have written a MINDSTORMS book or two. I remember manually stitching numerous screenshots together to get the right shots for larger programs. And then you decided to change your program and you had to start over. That won't be needed now :smile:

BertLindeman, 2024-01-18:
> That's right - that way you always get your screenshots in the same resolution, same margin, and no need to worry about forgetting to keep a block in view.
>
> Some of the people in this thread have [...]

I can tell (having those books) that person did well 😃

BeksOmega, 2024-01-24:
> I was thinking again about the idea of embedding the program into the PNG.

App Inventor has done this! Source code is here: https://github.com/mit-cml/appinventor-sources/blob/d89e86499142513d68b0acaf94dd1c237d06a2d5/appinventor/blocklyeditor/src/exportBlocksImage.js#L462

laurensvalk, 2024-01-24:
Thanks @BeksOmega!

laurensvalk, 2024-03-11:
A basic UI for this has been added and will be part of this week's stable release. Thanks everyone for your input!

closed 2024-03-11 by laurensvalk

BlueBillCanada, 2024-03-26:
Beautiful, thank you.