pybricks/support #1351: [Feature] Show loading animation for background updates.

laurensvalk, 2023-12-22:
**Is your feature request related to a problem? Please describe.**
We frequently get questions about certain new features not loading or not being available.

In many cases, it's because the user's app has not been updated yet. It is not really clear that an update is in progress, so the user can't really know this. 

**Describe the solution you'd like**
Since we load the new app in the background, I wonder if we could have some visual indication that this is ongoing/happening? Maybe like the pyodide load animation.

dlech, 2023-12-22:
If you click the "check for update" button it will either show a toast saying that the app is up to date or turn into an indeterminate progress indicator until the full update is downloaded.

laurensvalk, 2023-12-22:
I'm aware, but this is the part that most people don't seem to find and in some cases it doesn't seem to work (always says up to date). I haven't really been able to find what's causing it.

closed 2024-02-19 by laurensvalk