pybricks/support #1355: [Bug] Hub Refuses to Connect 50% of the time

WhyFenceCode, 2023-12-27:
**Describe the bug**
The hub just fails to connect around 20% of the time, and the rest of the time it works just fine. The hub is right beside my computer, and on the attempts that it works, the bluetooth light instantly goes blue, and on the ones were it fails, it just keeps flashing.

**To reproduce**
Steps to reproduce the behavior:
1. Press the bluetooth connect hub button.
2. Select your hub.
3. Wait for problems to occur.
4. See red connection failed error.

**Expected behavior**
I expected I higher rate of connections working, as once I get it connected, it is very stable.

dlech, 2023-12-27:
Some Bluetooth adapters work better than others. To hazard a guess, you are using Windows with a built-in Intel combo Wi-fi + Bluetooth?

WhyFenceCode, 2023-12-27:
True, some do work better than others, and I am on exactly what you described, the only reason I mention this is everything else I have works really well instantly.

dlech, 2023-12-27:
If you want to dig deeper, you can try logging Bluetooth packets as described in #270. Or if you want to try a different Bluetooth adapter, there are some know good ones listed in #921.