pybricks/support #1357: [bug] Remote light does not work well with async

laurensvalk, 2023-12-28:
> I also did install the latest nightly firmware build for the technic hub. But I always have problems when I use remote and async on the technic hub together.
This is just a simple program. The remote is connected just for a short time, and then it is disconnected without any error message.

[image]

> The same program is running on a move hub without any issues.

_Originally posted by @mahil in https://github.com/pybricks/support/issues/1281#issuecomment-1871433043_

laurensvalk, 2023-12-28:
Thanks for reporting this.

Writing values asynchronously remains a tricky issue on the Technic Hub. Maybe we should not make the remote hub light controllable with blocks to make this less of an issue.

The button state can still be read without issues (which is probably the most important part of the remote).

laurensvalk, 2023-12-28:
@mahil - does it work better if delete the remote light blocks (and keep the remote is pressed block)?

mahil, 2023-12-29:
> @mahil - does it work better if delete the remote light blocks (and keep the remote is pressed block)?

Yes, it works.

mahil, 2023-12-29:
I did another test. The lights shall only be sets once when the status has changed in the loop. But this test failed as well.

[image]

mahil, 2023-12-29:
> Thanks for reporting this.
>
> Writing values asynchronously remains a tricky issue on the Technic Hub. Maybe we should not make the remote hub light controllable with blocks to make this less of an [...]

I believe it would be better to keep the lights for the remote control, but do not allow multitask instead (when you are using a remote controller). Otherwise, you cannot send any feedback to the user.

My bug report was just an example that something is not working as expected. This example can run in a single task without any problems.

BertLindeman, 2024-01-12:
Maybe same situation in #1373.
Async and bluetooth combined crash the Technic hub.

laurensvalk, 2024-02-01:
The source code with the fix will be published within a few weeks alongside a fun new feature.

There were a few issues here. This was not related to async in the end. The data being written to the remote was not statically allocated, so it could be gone by the time it would send it to the remote. We've had this bug for a while, but it wasn't so noticeable without async which has a lot more going on in parallel.

Thanks again for reporting!

laurensvalk, 2024-02-19:
This has been fixed and the source has now been published, so we can close this issue.

Main fix via https://github.com/pybricks/pybricks-micropython/commit/8a47892cc52dbf23ad837e21665975a20cac2aac

closed 2024-02-19 by laurensvalk