pybricks/support #1368: Pybricks Connection Problem

sjlct123, 2024-01-04:
I want to connect my Lego Mindstorms 51515 Hub to pybricks and install the pybricks firmware but it dosnt work. I can't even install WinUSB driver. It glitches with error code 10.

I expected the firmware to be installed on my hub correctley.

[image]

dlech, 2024-01-05:
Does the device show up in the Windows *Device Manager* as "LEGO Technic Large Hub in DFU Mode"?

Do you have a different computer you could try?

sjlct123, 2024-01-05:
No it doesn't show up as Lego technic large hub in dfu mode and I don't have any other computers to try it on.

dlech, 2024-01-05:
Is the Bluetooth light on the hub blinking red (pink)/green/blue/off? If not, then the hub is not in DFU mode and that is the problem. If it is, I would try deleting the device from the Device Manager and start over.

sjlct123, 2024-01-05:
It was in dfu mode but it didn't show up.

dlech, 2024-01-06:
Can you share a screenshot of your *Device Manager*? There should be something in there with a :warning: icon if the driver is not loading correctly.

sjlct123, 2024-01-06:
[image]

dlech, 2024-01-06:
If you right-click that and select *Uninstall device* then disconnect the hub and plug it back in (making sure to hold the Bluetooth button to get DFU mode) what does it say?

sjlct123, 2024-01-07:
It dosen't even show up on device manager anymore.

dlech, 2024-01-07:
Maybe there is a bad USB cable or something like that?

sjlct123, 2024-01-08:
I tried a different cable but it still didn't work.

dlech, 2024-01-08:
On my Windows 11, if I uninstall the device like I mentioned in https://github.com/pybricks/support/issues/1368#issuecomment-1879778338 and then unplug the hub and plug it back in, I see this:

[image]

You are not seeing this after following the same steps?

sjlct123, 2024-01-08:
Yes.

sjlct123, 2024-01-08:
I don't think it's my computer. I tried it on another computer but it didn't work. Thank you.

laurensvalk, 2024-09-16:
This has come up in several other topics. In those cases, using a different (or the original) USB cable fixed it. Let's close this issue until this occurs again. Thanks!

closed 2024-09-16 by laurensvalk