pybricks/support #1379: run_until_stalled() stops working when adding args.

antonvh, 2024-01-14:
When I add a duty limit, the command starts running the motor in the wrong direction and fails to execute run_target() later in the program. Here's the code I used. See comments for expected and unexpected results.

I ran this on a Technic hub. Haven't tried on a prime or inventor hub yet.

``` python
from pybricks.pupdevices import Motor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.tools import wait, StopWatch

steer = Motor(Port.A, Direction.CLOCKWISE)
# This works. Motor starts Clockwise, 100 deg/s
steer.run_until_stalled(100)
print("stalled at: ", steer.angle()) # returns a positive number
steer.reset_angle(300)
print("reset angle at: ", steer.angle()) # returns 200, neatly
steer.run_target(100, 0)
# Motor runs as expected.
print("Ran to: ", steer.angle()) 
steer.close()

# Now the trouble begins.
steer = Motor(Port.A, Direction.CLOCKWISE)
# Expecting the motor to run clockwise, because 100>0. But motor starts CCW
steer.run_until_stalled(100, Stop.COAST, 10) # < the only difference are the params here
print("stalled at: ", steer.angle()) # returns a negative number !!?!
steer.reset_angle(200)
print("reset angle at: ", steer.angle()) # returns 200, neatly
steer.run_target(100, 0)
# returns a number between -5 and 7 BUT the motor did NOT move!
print("Ran to: ", steer.angle()) 
```

BertLindeman, 2024-01-14:
> Does this happen when you don't use `close()` too?

Took the close out and replaced with a ```wait(500)``` 
That results in:

```
Traceback (most recent call last):
  File "issue_1379.py", line 24, in <module>
OSError: [Errno 16] EBUSY: Device or resource busy
```

Line number may be different as I already changed some other things.

BertLindeman, 2024-01-14:
Anton,

In the first part of the program you have:
```
steer.reset_angle(300)
print("reset angle at: ", steer.angle()) # returns 200, neatly
```

Did you mean here 200 in stead of 300? 
The second part has ```200```

Running with ```steer.reset_angle(200)``` it looks OK to me.
Running after the reset to 300 to zero stalls (in my setup) to the pin I have there.

The picture shows the position after the first part has been run (with rest to 200), and also after the program ends.
[image]

Looks oke to me then.
Do I misinterpret something? I am good at that :wink:

laurensvalk, 2024-01-14:
> > Does this happen when you don't use `close()` too?
> 
> Took the close out and replaced with a `wait(500)` That results in:
> 
> ```
> Traceback (most recent call last):
>   File "issue_1379.py", line 24, in <module>
> OSError: [Errno 16] EBUSY: Device or resource busy
> ```

That is expected. I just meant to ask if the `close` combined with the second `Motor` initialization could be contributing to this issue.

antonvh, 2024-01-14:
> Does this happen when you don't use `close()` too?

Yes. I only used close() to put it all in one program to make it easy to reproduce. Originally, it was separate programs.

antonvh, 2024-01-14:
> n the first part of the program you have:
> 
> ```
> steer.reset_angle(300)
> print("reset angle at: ", steer.angle()) # returns 200, neatly
> ```
> 
> Did you mean here 200 in stead of 300? The second part has `200`
Yes. It was a typo. It needs to be 200 everywhere. But the bug persists regardless.

> Running with `steer.reset_angle(200)` it looks OK to me. Running after the reset to 300 to zero stalls (in my setup) to the pin I have there.
> 
> The picture shows the position after the first part has been run (with rest to 200), and also after the program ends.
That it fails with 300 is also to be expected, as there is no room in your setup to accommodate that.

However, I replicated your setup, with both an angular technic motor and a medium technic motor (non angular). I then used reset_angle(100) everywhere. What is strange is that the problem is intermittenet. Sometimes it works as expected, but the next time I run the program it fails and the motor starts the wrong way. Failure happens more often with the angular motor.

BertLindeman, 2024-01-14:
> However, I replicated your setup, with both an angular technic motor and a medium technic motor (non angular). I then used reset_angle(100) everywhere. What is strange is that the problem is [...]

Thanks for your feedback, will look at it with the angular motor.

BertLindeman, 2024-01-14:
Anton, I can confirm that the same program acts differently on different motors.
No at all sure which motors exactly. 
Also did not test on different hubs.

BertLindeman, 2024-01-14:
A weird idea.

The various motors have different characteristics.
Could it be that a duty-cycle of 10 on the stall is **too low** compared to the internal friction of the various motors?

BertLindeman, 2024-01-17:
@antonvh Did you have time to look at the previous post? 
Forgot that a post update does not send a notification.

Bert

antonvh, 2024-01-17:
I will also make a movie later. 'Fake' stall is not an issue for me. It would be expected behavior. The real problem is that the motor starts in the wrong direction and refuses to execute run_target() afterwards.

BertLindeman, 2024-01-17:
> I will also make a movie later. 'Fake' stall is not an issue for me. It would be expected behavior. The real problem is that the motor starts in the wrong direction and refuses to execute run_target() afterwards.

Thank you Anton, will await your movie.

laurensvalk, 2024-02-27:
This is working correctly. 

The confusion probably arises from `duty_limit=10`, which means the motor won't move at all and stall right away. See notes inline below.

```
# Now the trouble begins.
steer = Motor(Port.A, Direction.CLOCKWISE)
# Expecting the motor to run clockwise, because 100>0. But motor starts CCW                 # See * below
steer.run_until_stalled(100, Stop.COAST, 10) # < the only difference are the params here    # 10 % is not enough to move the motor, so it stall right away
print("stalled at: ", steer.angle()) # returns a negative number !!?!                       # Yes. When initializing the motor it starts at the absolute position (-180 to 180)
steer.reset_angle(200)                                                                      
print("reset angle at: ", steer.angle()) # returns 200, neatly                              # Correct.
steer.run_target(100, 0)                                                                    # * and this is the first movement. We go from 200 to 0, so CCW.
# returns a number between -5 and 7 BUT the motor did NOT move!                             # But it did move, see above. Correct the default error margin will be a few degrees around 0.
print("Ran to: ", steer.angle()) 
```

closed 2024-02-27 by laurensvalk