pybricks/support #1381: [Bug] New USB library prevents checkout of past commits

laurensvalk, 2024-01-16:
Adding and removing submodules is always a bit cumbersome, but usually checkout at least works. Now it doesn't seem to.

```
cd pybricks-micropython
git checkout master
make primehub -j
git checkout v3.3.0
```

```
error: The following untracked working tree files would be overwritten by checkout:
        lib/STM32_USB_Device_Library/Class/AUDIO/Inc/usbd_audio.h
        lib/STM32_USB_Device_Library/Class/AUDIO/Inc/usbd_audio_if_template.h
        lib/STM32_USB_Device_Library/Class/AUDIO/Src/usbd_audio.c
        lib/STM32_USB_Device_Library/Class/AUDIO/Src/usbd_audio_if_template.c
        lib/STM32_USB_Device_Library/Class/BillBoard/Inc/usbd_billboard.h
        lib/STM32_USB_Device_Library/Class/BillBoard/Src/usbd_billboard.c
        lib/STM32_USB_Device_Library/Class/CDC/Inc/usbd_cdc.h
        lib/STM32_USB_Device_Library/Class/CDC/Inc/usbd_cdc_if_template.h
        lib/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc.c
        lib/STM32_USB_Device_Library/Class/CDC/Src/usbd_cdc_if_template.c
        lib/STM32_USB_Device_Library/Class/CDC_ECM/Inc/usbd_cdc_ecm.h
        lib/STM32_USB_Device_Library/Class/CDC_ECM/Inc/usbd_cdc_ecm_if_template.h
        lib/STM32_USB_Device_Library/Class/CDC_ECM/Src/usbd_cdc_ecm.c
        lib/STM32_USB_Device_Library/Class/CDC_ECM/Src/usbd_cdc_ecm_if_template.c
        lib/STM32_USB_Device_Library/Class/CDC_RNDIS/Inc/usbd_cdc_rndis.h
        lib/STM32_USB_Device_Library/Class/CDC_RNDIS/Inc/usbd_cdc_rndis_if_template.h
        lib/STM32_USB_Device_Library/Class/CDC_RNDIS/Src/usbd_cdc_rndis.c
        lib/STM32_USB_Device_Library/Class/CDC_RNDIS/Src/usbd_cdc_rndis_if_template.c
        lib/STM32_USB_Device_Library/Class/CustomHID/Inc/usbd_customhid.h
        lib/STM32_USB_Device_Library/Class/CustomHID/Inc/usbd_customhid_if_template.h
        lib/STM32_USB_Device_Library/Class/CustomHID/Src/usbd_customhid.c
        lib/STM32_USB_Device_Library/Class/CustomHID/Src/usbd_customhid_if_template.c
        lib/STM32_USB_Device_Library/Class/DFU/Inc/usbd_dfu.h
        lib/STM32_USB_Device_Library/Class/DFU/Inc/usbd_dfu_media_template.h
        lib/STM32_USB_Device_Library/Class/DFU/Src/usbd_dfu.c
        lib/STM32_USB_Device_Library/Class/DFU/Src/usbd_dfu_media_template.c
        lib/STM32_USB_Device_Library/Class/HID/Inc/usbd_hid.h
        lib/STM32_USB_Device_Library/Class/HID/Src/usbd_hid.c
        lib/STM32_USB_Device_Library/Class/MSC/Inc/usbd_msc.h
        lib/STM32_USB_Device_Library/Class/MSC/Inc/usbd_msc_bot.h
        lib/STM32_USB_Device_Library/Class/MSC/Inc/usbd_msc_data.h
        lib/STM32_USB_Device_Library/Class/MSC/Inc/usbd_msc_scsi.h
        lib/STM32_USB_Device_Library/Class/MSC/Inc/usbd_msc_storage_template.h
        lib/STM32_USB_Device_Library/Class/MSC/Src/usbd_msc.c
        lib/STM32_USB_Device_Library/Class/MSC/Src/usbd_msc_bot.c
        lib/STM32_USB_Device_Library/Class/MSC/Src/usbd_msc_data.c
        lib/STM32_USB_Device_Library/Class/MSC/Src/usbd_msc_scsi.c
        lib/STM32_USB_Device_Library/Class/MSC/Src/usbd_msc_storage_template.c
        lib/STM32_USB_Device_Library/Class/Template/Inc/usbd_template.h
        lib/STM32_USB_Device_Library/Class/Template/Src/usbd_template.c
        lib/STM32_USB_Device_Library/Core/Inc/usbd_conf_template.h
        lib/STM32_USB_Device_Library/Core/Inc/usbd_core.h
        lib/STM32_USB_Device_Library/Core/Inc/usbd_ctlreq.h
        lib/STM32_USB_Device_Library/Core/Inc/usbd_def.h
        lib/STM32_USB_Device_Library/Core/Inc/usbd_desc_template.h
        lib/STM32_USB_Device_Library/Core/Inc/usbd_ioreq.h
        lib/STM32_USB_Device_Library/Core/Src/usbd_conf_template.c
        lib/STM32_USB_Device_Library/Core/Src/usbd_core.c
        lib/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c
        lib/STM32_USB_Device_Library/Core/Src/usbd_desc_template.c
        lib/STM32_USB_Device_Library/Core/Src/usbd_ioreq.c
        lib/STM32_USB_Device_Library/Release_Notes.html
Please move or remove them before you switch branches.
```

Workaround:

```
rm -rf lib/STM32_USB_Device_Library
```

dlech, 2024-01-21:
`git checkout -f master` works for me.

laurensvalk, 2024-01-21:
It's checking out anything before the library change, e.g. `v3.3.0` that seems to be problematic. Probably not much we can do about it now though.

closed 2024-02-27 by laurensvalk