pybricks/support #1388: [Bug] LEGO SPIKE will suddenly power off during the robot running.

a10036gt, 2024-01-21:
LEGO SPIKE will suddenly power off during the robot running.

We are praticing the FLL, and sometimes when robot call DriveBase turn function (with use_gyro() enable), SPIKE will suddenly power off during the robot running, the battery is fixed well on our robot and voltage is 8.1v, so maybe not the battery issue.

It is a very rare situation. It may take an entire afternoon to run and it will only occur once. I am still observing it after changing other SPIKE.

v3.3.0 (Pybricks Code v2.3.0)

laurensvalk, 2024-01-21:
Thanks for reporting this. Can you share a program snippet that has this issue?

The smaller the program, the better. Thanks!

a10036gt, 2024-01-21:
It seems the suddenly power off occured when use gyro (DriveBase), this afternoon happen 3 times.
[image]

I am still observing it after changing other SPIKE Hub.

laurensvalk, 2024-01-21:
Does the program stop, or does the hub power off?

If you run it while it is still connected in the app, does it print any error messages?

a10036gt, 2024-01-21:
I am sure the hub is power off, because when suddenly poweroff and next time it turn on with matrix light (the square) animation.
When we run program the bluetooth is still connect to app, and there is no any error messages on it.

laurensvalk, 2024-01-21:
It sounds like it could be the battery.

In SPIKE Prime, the battery is often quite loose in the hub.

You can see this for yourself - you can probably move the battery around inside the hub.

And this is more likely to happen while it is moving around, so maybe this is why you saw it happen while driving?

Some teams have solved this by putting a piece of cardboard on the other side to keep it in place.

If this is the case, it's perhaps not an issue with Pybricks.

a10036gt, 2024-01-21:
So even my hub is quite fits the motor very well, the battery still possible loose in the hub right?
[image]

Maybe I should putting a piece of cardboard into it keep it in place. It's my frist time to use SPIKE and PyBricks, we using EV3 before, thanks to PyBricks teams bring the very great software for us! 

Is there any plan to add regular UART or I2C for PyBricks SPIKE/RI support, it will be very useful when we want to add some custom sensors like Pixy2, now we need use arduino or stm8 to emulate as LUMP device to create the bridge between 3rd party sensors and Hub.

Btw, the simple robot is driving square for 2h30m now, seems fine.

laurensvalk, 2024-01-21:
> So even my hub is quite fits the motor very well, the battery still possible loose in the hub right?

Right, the movement is often sideways inside the hub. The easiest way to see it, is if you take it out of the robot.

> It's my frist time to use SPIKE and PyBricks, we using EV3 before, thanks to PyBricks teams bring the very great software for us!
> (...)
> Btw, the simple robot is driving square for 3h30m now, seems fine.

That's awesome :smile: 

> Is there any plan to add regular UART or I2C for PyBricks SPIKE/RI support, it will be very useful when we want to add some custom sensors like Pixy2, now we need use arduino or stm8 to emulate as LUMP device to create the bridge between 3rd party sensors and Hub

While we have no short term plans for this, it's on our list for the long term :)

a10036gt, 2024-01-21:
Thanks for reply😄

The test robot is turn off because run out of battery (6.0v), 4hrs ABCD motor full load with 2 color sensor, so I think the problem might be battery loose, if have any issues or discover I will update.

closed 2024-01-21 by a10036gt

a10036gt, 2024-02-03:
Ok let me update some info here, since we ask students putting the robot carefully and we add some tape behind the battery, the problem seems gone, so I am quite sure it's battery loose issue.

And thanks to the Pybricks provide really great software solutions, we got the 2nd place in FLL Taiwan National tournament, Pybricks works really fast and stable! The link below is our Robot Game video whitch is code with Pybricks!🤩

https://youtu.be/q1MxhqVm4yY?si=QOm88KWp321jIBQd

And I really hope can have Regular I2C or UART that can work with Pixy camera for WRO😉

laurensvalk, 2024-02-03:
Thanks for the update!

Well done to your team, that's truly amazing! 

Awesome video too. Thanks for taking the time to record.