pybricks/support #1389: [Question] WinUsb not listed in available derivers list

SDPrio, 2024-01-21:
My question is basically the same as in issue #1248, however no solution was posted there.

I am using an up to date Windows 11 and tried to follow the instructions to install the WinUsb driver (e.g. in issue #1067). This is no problem up to the step no. 6 where "WinUsb Device" is not included in the list of manufactures.

As in issue #1248 I see a long list of manufactures, but neither "Lego Education" nor "WinUsb Device" is listed here. 

A comment in issue #1248 indicated that one has to select "Microsoft" first. However, this only shows some entries in the "Model" box where I can select between Microsoft Keyboards, etc. 

So, where to find the WinUsb Device entry? Maybe the manually should be updated to show this in more detail.

laurensvalk, 2024-01-21:
Thank you. I don't have a Windows 11 machine, so I'm not exactly sure. Can you share a screenshot of the final step that still matches the intructions, and then also a screenshot of what you see when there is no match?

SDPrio, 2024-01-21:
[image]

On the left you can the screenshot from the instructions where "WinUsb" is listed under "Manufacture". On the right you can see my list of manufactures. While "Microsoft" is included, "WinUsb" isn't

laurensvalk, 2024-01-21:
Thanks. And I think you're saying there also isn't something like WinUSB or WindowsUSB device on the right hand side list if you choose Microsoft?

By the way, which devices do you see if you choose Lego Education?

SDPrio, 2024-01-21:
OK, it seeps that in step 5 of the instructions the the "Universal Serial Bus devices" entry is named differently in my German windows version. After trying some other entries in this step I finally got to the list with "WinUsb". 

Thanks!

laurensvalk, 2024-01-21:
Let's keep this issue open as a reminder so we can fix the screenshots this time :smile: 

Would you mind sharing the exact names / lists you found? Even if it is in German, that's already a good help.

laurensvalk, 2024-01-25:
Hi @SDPrio - what was the solution for you?

It would be nice to know, as it may benefit other users.

SDPrio, 2024-01-25:
Oh, I am sorry. I have not noticed, that the topic had been reopened. 

Actually I helped my nephew to setup the project when I visited him. Now I do not have a Windows PC to check the actual translations. I will try to create a screenshot next time I have a chance.

laurensvalk, 2024-02-19:
It does appear to be available in Window 11, but I think the lists are a bit longer than in Windows 10, which can make it look quite different, especially in other languages.

BertLindeman, 2025-01-25:
Some loose hints:

I am **not sure**, but I think the hub should **not** be connected / paired to windows.

I see no problem with this on my (US-English) Windows 11 at 24H2.
You may need to install the winusb driver for **each** Spike / Robot inventor hub.

On this windows pc this driver shows up like this:

[image]

---

#### 1. **Verify USB Connection**
- **Check the USB cable:** Make sure you’re using a **data cable** (not just a charging cable). Some USB cables only carry power and won’t transfer data.
- **Try different ports:** Use a direct USB port instead of a USB hub or adapter.

---

#### 2. **Confirm Bootloader Mode**
- The **3-color blinking on the Bluetooth button** indicates the hub is in bootloader mode.
- Bootloader devices might appear under different categories in **Device Manager**, such as:
  - **Universal Serial Bus Devices** (USB-Geräte)
  - **Ports (COM & LPT)** (Anschlüsse)
  - **Unknown Devices** (Unbekannte Geräte)

If you don’t see the hub in any of these, move to the next steps.

---

#### 3. **Refresh Device Manager**
Sometimes the device doesn’t immediately show up in **Device Manager**. 
Try the following:
1. In **Device Manager**.
2. In the menu, click **Action** → **Scan for hardware changes** (or **Nach geänderter Hardware suchen**).
3. Look for any new or unknown devices.
4. If the hub still isn’t visible:
 - In the menu, click **View** → **Show hidden devices** (Ausgeblendete Geräte anzeigen).
 - Check for grayed-out entries or an unrecognized device.

---

#### 4. **Test on Another PC**
If the hub still doesn’t appear, test it on another computer. If it’s recognized there, the issue may be with your current machine's USB drivers or settings.

---

Bert

dlech, 2025-04-20:
I think this is the same issue as https://github.com/orgs/pybricks/discussions/2032. The problem is not on the page in the screenshots above, but rather on the page before that. In other languages, it isn't always clear the difference between USB "devices" and USB "controllers".  These both have the same icon, so it is easy to click the wrong one. WinUSB is only listed under drivers for USB "devices".

[image]