pybricks/support #1391: [Bug] Non-cancellable bluetooth tasks prevent shutdown

laurensvalk, 2024-01-21:
This issue has gone through various iterations over time, so I'd need to go back through the history to see how it used to work exactly. There is currently this note, but I don't think it's currently working in the `cc2640` driver.

```c
void pbsys_program_stop(bool force_stop) {
    if (pbsys_status_test(PBIO_PYBRICKS_STATUS_USER_PROGRAM_RUNNING)) {
        if (force_stop) {
            // HACK: Bluetooth drivers have issues with pending tasks not being
            // cancellable, so this should force them to be canceled and avoid
            // undefined behavior due to use after free.
            pbdrv_bluetooth_power_on(false);
        }
        pbsys_main_stop_program(force_stop);
    }
}
```

**To reproduce**
Todo.

Basically, any non-cancelable wait on some event.

We should ideally avoid this, but it would be better if we can avoid having to pull the batteries if it does happen. Or such that if the event does eventually arrive and the user has already tried to shut the hub down, that we don't stay stuck.

I thought we used to have a system abort that would always be able to exit the program at least on shutdown, but it doesn't seem to be working now.

laurensvalk, 2024-01-25:
Workaround / fix implemented via https://github.com/pybricks/pybricks-micropython/commit/e836b04fafa193bef6a9f8065f63d2ad22af63cd

dlech, 2024-01-25:
This fix seems dangerous since it will leave the hub in an unknown state. I would only do this if the hub is shutting down. Otherwise, the hub is likely to crash soon.

BertLindeman, 2024-01-25:
Might this issue have anything to do with a Technichub at build 3232 in a rapid blinking state and cannot be stopped by a long button press?

laurensvalk, 2024-01-25:
> This fix seems dangerous since it will leave the hub in an unknown state. I would only do this if the hub is shutting down. Otherwise, the hub is likely to crash soon.

True, but a forced reset is probably still slightly better than just hanging in the state Bert described above.

I'll see if we can bring back the forced shutdown that we used to have, instead of this.

I'd also like to review the cases where `tasks` are used, and whether we could use static singletons to avoid use-after-free risks. The most important ones are already static, I think, and we could rewrite the dynamic ones since we already don't allow simultaneous calls to Bluetooth anyway.

laurensvalk, 2024-01-26:
> I'd also like to review the cases where tasks are used, and whether we could use static singletons to avoid use-after-free risks. 

Only the `hub.ble` task is dynamically allocated, but we can use a static singleton for it since there is only one hub anyway.

laurensvalk, 2024-01-26:
https://github.com/pybricks/pybricks-micropython/commit/4d386be000dd4b49fc764ba80c23a95e43d219b5 and the FIXME introduced there seems important:

```
    // FIXME: need a way to poll the process that has the task queue to
    // expedite the cancellation. e.g. if the task is not the first in the
    // queue, it can just be removed from the queue. For btstack, we probably
    // need to create a synthetic event. For other drivers, we can probably
    // do something with contiki events. Since it is driver-specific, this
    // means we probably need to add a cancel callback to the task struct.
```
On cancellation, tasks don't actually run again so hanging is probably expected.

laurensvalk, 2024-01-26:
I will try to create a pull request for some ongoing Bluetooth updates ASAP.

We can probably come a long way by improving the existing cancels another way. Some of them cancel fine.

dlech, 2024-01-26:
> and whether we could use static singletons to avoid use-after-free risks.

It would avoid corrupting the stack, but the static struct could still be written over by another call if cancellation isn't handled properly. So would still have undefined behavior (although probably less bad that corrupting the stack). So really should reboot at this point anyway.

laurensvalk, 2024-01-26:
I was able to avoid needing either solution for the pairing step (which can now be properly cancelled), so the need for this is just back to normal :smile: (but still good to clean up).

laurensvalk, 2024-02-19:
This was properly fixed via https://github.com/pybricks/pybricks-micropython/pull/234 instead.

closed 2024-02-19 by laurensvalk