pybricks/support #1397: [Feature] New Car class - Feedback wanted!

laurensvalk, 2024-01-25:
**Is your feature request related to a problem? Please describe.**
Almost all (Technic) vehicles work in the same way:
- One or more drive motors.
- A steering motor that needs to initialize to find the center.

In a way, this type of build is just another abstraction of motors, just like the `DriveBase`, so it makes sense to have a dedicated class for it.

**Proposed API, example code**

```python
from pybricks.parameters import Direction, Port, Button
from pybricks.pupdevices import Motor, Remote
from pybricks.robotics import Car

# Set up motors.
front = Motor(Port.A, Direction.COUNTERCLOCKWISE)
rear = Motor(Port.B, Direction.COUNTERCLOCKWISE)
steer = Motor(Port.C, Direction.CLOCKWISE)

# Connect to the remote
remote = Remote()

# Set up the car.
car = Car(steer, [front, rear])

# The main program starts here.
while True:

    # Read remote state.
    pressed = remote.buttons.pressed()

    # Steer using the left pad. Steering is in percentage of the angle
    # determined while initializing.
    steering = 100 if Button.LEFT_PLUS in pressed else (-100 if Button.LEFT_MINUS in pressed else 0)
    car.steer(steering)

    # Drive using the left pad.
    power = 100 if Button.RIGHT_PLUS in pressed else (-100 if Button.RIGHT_MINUS in pressed else 0)
    car.drive_power(power)
```

# Details

This is one of our first features implemented in Python instead of C, since it wasn't going to fit on Move Hub anymore anyway.

You can view the implementation [here](https://github.com/pybricks/pybricks-micropython/blob/e5bd9a4de44675a3870111f9cd701d94aa20ce9a/bricks/_common/modules/_robotics_extra.py).

# Try it out

[See instructions for testing the latest build.](https://pybricks.com/install/spike-mindstorms/#installing-the-latest-build-advanced)

# Questions for you

Is this a useful class? Are any methods missing? Do the parameters make sense? Anything else we should reconsider before releasing it?

laurensvalk, 2024-01-25:
Also tagging @tensafefrogs since you have recently submitted several pull requests for such vehicles.

BertLindeman, 2024-01-25:
Tried technic hub with 3 motors (not a real car, but just loose motors)
Works great.
Although I would have named the drive motors maybe left and right.

Tested cityhub, I wondered if one drive motor would do the trick. 
Sure does.

So easy to have a vehicle up and running.
Do we account for the difference in the rotation of the wheels during a turn?

laurensvalk, 2024-01-25:
Thanks for trying it!

The motors are named that way in this example, but they're just the `drive_motors` in the class.

> Although I would have named the drive motors maybe left and right.
> (...)
> Do we account for the difference in the rotation of the wheels during a turn?

They are often powering the front and rear wheels in models such as the [off roader](https://pybricks.com/projects/sets/technic/42099-off-roader/), where a mechanical differential takes care of left and right.

laurensvalk, 2024-01-25:
Block version becomes this simple!

Maybe the ternary blocks are a bit much (normal switch blocks are fine), but it does become much simpler than before.

[image]

BertLindeman, 2024-01-25:
I mis-understood your remark:
> since it wasn't going to fit on Move Hub anymore anyway.

Assumed this python code would be provided for the movehub.
But ```Car``` is not available on the movehub.

Really like the blocks program.

tensafefrogs, 2024-01-25:
This looks great and would save a bunch of common code. Unfortunately I can't test it because my kid took apart our etron 😅

I did ask chat gpt to try it out using the etron code and it looks much cleaner:

`from pybricks.pupdevices import Motor, Remote
from pybricks.parameters import Port, Direction, Button
from pybricks.robotics import Car
from pybricks.tools import wait, StopWatch
import pybricks.hubs

# Initialize the motors and Car class
steer = Motor(Port.D, Direction.CLOCKWISE)
front = Motor(Port.B, Direction.COUNTERCLOCKWISE)
rear = Motor(Port.A, Direction.COUNTERCLOCKWISE)
car = Car(steer, [front, rear])

# Connect to the remote
remote = Remote()

# Speed levels and current speed index
speed_levels = [500, 1000, 2000]  # Slow, Medium, Fastest
current_speed_index = 1  # Start with Medium speed

# Variables to track the last state of the red buttons
prev_left_red_pressed = False
prev_right_red_pressed = False

# Initialize inactivity timer
inactivity_timer = StopWatch()

while True:
    pressed = remote.buttons.pressed()
    
    # Reset inactivity timer on button press
    if pressed:
        inactivity_timer.reset()

    # Shutdown on inactivity
    if inactivity_timer.time() > 120000:  # 2 minutes in milliseconds
        pybricks.hubs.TechnicHub.system.shutdown()

    # Adjust speed index based on red button presses
    if Button.LEFT in pressed and not prev_left_red_pressed:
        current_speed_index = max(0, current_speed_index - 1)  # Decrease speed index
    if Button.RIGHT in pressed and not prev_right_red_pressed:
        current_speed_index = min(len(speed_levels) - 1, current_speed_index + 1)  # Increase speed index

    # Apply the current speed level
    power = speed_levels[current_speed_index] if Button.LEFT_PLUS in pressed else (-speed_levels[current_speed_index] if Button.LEFT_MINUS in pressed else 0)
    car.drive_power(power)

    # Control steering
    steering = 100 if Button.RIGHT_PLUS in pressed else (-100 if Button.RIGHT_MINUS in pressed else 0)
    car.steer(steering)

    # Update the previous button states
    prev_left_red_pressed = Button.LEFT in pressed
    prev_right_red_pressed = Button.RIGHT in pressed

    wait(25) 
`

tensafefrogs, 2024-01-25:
Doh sorry, code formatting on my phone is too hard, feel free to edit that or I will later :)

closed 2024-02-19 by laurensvalk