pybricks/support #1410: [Bug] unpack block does not set globals correctly

laurensvalk, 2024-02-02:
> Broadcasting and receiving values work fine, however, the variables won't update in the entire program at once. See my simple program attached. I have two parallell programs running at different intervals. One hub sends the "TriggerDistributor" value (true) by broadcasting. This value won't be changed for about three seconds or so, and then go back to being false. This is picked up by the first print function and shows in the console, however the other print function in the lower program will always print false. This is a problem since I'm using these values in different parts of the program, usually with "wait until" blocks, waiting until the variable becomes true.

[image]

_Originally posted by @JJHackimoto in https://github.com/pybricks/support/discussions/1373#discussioncomment-8339692_

laurensvalk, 2024-02-02:
This is the correct behavior:

[image]

But when we use multitasking with async, it packs that stack of blocks into a function. So to set variables, it has to make them global, but it doesn't:

[image]

laurensvalk, 2024-02-02:
This is better! Fix should be coming soon...

[image]

laurensvalk, 2024-02-02:
Fix is out. Let me know how it goes :sunglasses: 

[image]

BertLindeman, 2024-02-02:
> Fix is out. Let me know how it goes 😎

Takes more time, obviously, after "check for update" and "Already up to date":

```
About Pybricks Beta
Pybricks logo
MicroPython for LEGO® Powered Up smart hubs.

v3.3.0 (Pybricks Beta v2.4.0-beta.1)

Copyright (c) 2020-2023 The Pybricks Authors

LEGO® is a trademark of the LEGO Group of companies which does not sponsor, authorize or endorse this site.
```

JJHackimoto, 2024-02-02:
> Fix is out. Let me know how it goes 😎 [...]

Worked perfectly! Thank you for submitting the report and fixing it so quickly!

laurensvalk, 2024-02-02:
Glad to hear it. And thank you for raising this. This was easily fixed, but would cause plenty of confusion in multi-tasked broadcasting, which is complex enough when it is working correctly :smile:

closed 2024-02-02 by laurensvalk