pybricks/support #1415: [Feature] Keep the hub name as a default when installing/re-sinatlling PyBricks Firmware

afarago, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
When flashing the hub with the pybricks firmware it would be awesome to keep the hub name intact.

**Describe the solution you'd like**
We have multiple hubs for robotics (FLL, WRO, Edutus Robotics Cup) and while flashing / updating the firmware it is cumbersome to re-enter the hub name again and again.

Possible solutions:
1) if connected to a hub and starting the "Install Pybricks Firmware" - take the name of the hub Instead of the default to the step3 - Hub name.
2) while flashing the firmware add an option (checkbox) to step3 - Hub name (optional) to keep the current name and read it when selecting the DFU in step 4.

I am not sure if hub name is available in DFU mode of a PyBricks firmware Hub.
I am not sure if hub name is available in DFU mode of a stock LEGO SPIKE or Mindstorms Hub.

**Describe alternatives you've considered**
Currently I check which hub I am flashing and re-entering the name.

**Additional context**
N/A

dlech, 2024-02-04:
> I am not sure if hub name is available in DFU mode of a PyBricks firmware Hub.

Sort of, but it isn't in a fixed location, so we currently don't have a way to find it. But the firmware could be modified to add some mechanism to find it.

> I am not sure if hub name is available in DFU mode of a stock LEGO SPIKE or Mindstorms Hub.

IIRC, there is a pointer table at a fixed position in the flash memory (at least for 2.x firmware, not sure about newer LE SPIKE 3 firmware) that has a pointer to the hub name, so maybe possible.

> When flashing the hub with the pybricks firmware it would be awesome to keep the hub name intact.

This is something I've thought about a time or two as well. The best possible solution I have though of so far would be to update the "default" hub name with the name of the last connected hub. For people with one hub, this would always be the right name. For people updating multiple hubs, they would have to connect regularly first to have the name read (no other interaction would be required, just connect) and then put the hub in update mode and go through the update process.

afarago, 2024-02-04:
> The best possible solution I have though of so far would be to update the "default" hub name with the name of the last connected hub
Would love this solution!