pybricks/support #1429: [Bug] await wait(0) does not yield (enough)

laurensvalk, 2024-02-08:
**Describe the bug**
The following program does not run the second stack of program, i.e. the buttons won't operate the gripper.

[image]

**To reproduce**

```python
from pybricks.hubs import InventorHub
from pybricks.parameters import Button, Direction, Port, Stop
from pybricks.pupdevices import Motor, Remote
from pybricks.tools import multitask, run_task, wait

# Set up all devices.
inventor_hub = InventorHub()
gripper = Motor(Port.E, Direction.CLOCKWISE)
remote = Remote(timeout=None)

async def main1():
    print('Hello, Pybricks!')
    while True:
        await wait(0) # < ---- automatically inserted, but not enough

async def main2():
    while True:
        await wait(0)
        if Button.LEFT in remote.buttons.pressed():
            await gripper.run_target(500, 80)
        elif Button.RIGHT in remote.buttons.pressed():
            await gripper.run_target(500, 0)
        else:
            await wait(50)

async def main():
    await multitask(main1(), main2())

run_task(main())
```

**Expected behavior**
Bad loops are caught by the block language to automatically yield. This is working, but apparently `await wait(0)` does not currently appear to yield at least once when it should.

The workaround would be to insert `wait(1)` but it would be better to fix it in the firmware.

laurensvalk, 2024-02-19:
So this program will not print anything.

```python
# pybricks blocks file:{"blocks":{"languageVersion":0,"blocks":[{"type":"blockGlobalSetup","id":"bjK,wS1MYO7aiYkFSwd{","x":0,"y":0,"deletable":false},{"type":"blockGlobalStart","id":"3tJe|AWl0baN(wH9a$@.","x":0,"y":310,"deletable":false,"next":{"block":{"type":"blockFlowWhile","id":"Y`O(Q.|.}LO5^LE_n#A1","fields":{"MODE":"WHILE"},"inputs":{"BOOL":{"shadow":{"type":"blockLogicTrue","id":")wL27xyv+6Wry5CKMTX]"}},"DO":{"block":{"type":"blockPrint","id":"j,,T}?rBkaW$1v?olp4p","extraState":{"optionLevel":0},"inputs":{"TEXT0":{"shadow":{"type":"text","id":"!x5.0YiWya^`(y)yO5B8","fields":{"TEXT":"Hello!"}}}},"next":{"block":{"type":"blockWaitTime","id":"/N%9G}=;hA+Uy=s2Jl5=","inputs":{"VALUE0":{"shadow":{"type":"unit_time","id":"#{I,Ja(L1mag!CXhBc)V","fields":{"VALUE0":500}}}}}}}}}}}},{"type":"blockGlobalStart","id":"aQN9ElJZ(m|j-td;SM:E","x":0,"y":98,"next":{"block":{"type":"blockFlowWhile","id":"@MO(=;?,Ed11r0TS.W|z","fields":{"MODE":"WHILE"},"inputs":{"BOOL":{"shadow":{"type":"blockLogicTrue","id":"M1lIh`uuwKr/7jLbuF_6"}}}}}}]},"variables":[{"name":"red","id":"k616.|pE7d$1Asq`.F5,","type":"ColorDef"},{"name":"orange","id":"8K*P2H7ZzSLLq/.-TU,{","type":"ColorDef"},{"name":"yellow","id":"$}$K5?=m_KZncelINtSC","type":"ColorDef"},{"name":"green","id":"62Hz2EHxfOONf^Tf$F;X","type":"ColorDef"},{"name":"cyan","id":":?-!+PJsRwP,C5*kER|]","type":"ColorDef"},{"name":"blue","id":"g-mtr8NWtBk(5R{}qo4(","type":"ColorDef"},{"name":"violet","id":"6@h=eP7n`OvDd[-]ZsJf","type":"ColorDef"},{"name":"magenta","id":"t2$g@QMI}t*s=:Z_YN?Q","type":"ColorDef"},{"name":"white","id":"XUnog]1HS`F6IzcHwy=n","type":"ColorDef"},{"name":"none","id":"!FqL91v%Y9V,f`)%tQ,@","type":"ColorDef"}]}
from pybricks.tools import multitask, run_task, wait

async def main1():
    while True:
        await wait(0)

async def main2():
    while True:
        await wait(0)
        print('Hello!')
        await wait(500)

async def main():
    await multitask(main1(), main2())

run_task(main())
```

[image]

laurensvalk, 2024-02-19:
We can simplify this and rule out the `multitask` function as the culprit because it still fails without it:

```python
from pybricks.tools import run_task, wait

async def main1():
    while True:
        await wait(0) # second taks only proceeds with nonzero delay

async def main2():
    while True:
        await wait(0)
        print('Hello!')
        await wait(500)

async def main():
    
    first = main1()
    second = main2()

    while True:
        next(first)
        next(second)

run_task(main())
```

laurensvalk, 2024-02-19:
So `await wait(0)` just doesn't yield.

So we could either
1. Make it yield once
2. Use `await wait(1)`

I kind of want to avoid `2` because it would mean simple math loops would unnecessarily take `1` ms around the loop.

But when looking at this, the loop time doesn't seem to work as intended which is a separate issue.

dlech, 2024-02-19:
My gut reaction here is to make `await wait(0)` yield once (pretty sure this is what `await asyncio.sleep(0)` does) and do https://github.com/pybricks/support/issues/1460#issuecomment-1952075493.

laurensvalk, 2024-08-26:
Fixes suggested in https://github.com/pybricks/pybricks-micropython/pull/262, https://github.com/pybricks/pybricks-micropython/pull/263.

While I was thinking about this, it occurred to me we could extend this by making all awaitables yield at least once. For example `await sensor.reflection()` now only yields if there is a mode change to be done. By making it always yield once, any loop with at least one `await` statement would be safe. But since Python doesn't do this, I think it is probably not good to teach this pattern, and so we shouldn't do this. And it would slow down all those statements by one extra iteration. End of mental note.