pybricks/support #1430: [Bug] Block undo operation can delete device block

laurensvalk, 2024-02-08:
**Describe the bug**

Undoing a change in device type can leave a device getter without a block:

[image]

**To reproduce**
TBD. Most likely, select one device (Color Distance Sensor), change the measurement read to another device (Color Sensor),  and then delete the color sensor setup block so it no longer exists. Then try to undo things.

BertLindeman, 2024-02-08:
Is this the situation you meant after some redo's?
[image]

I noticed that undo [CTRL-Z} allows more steps than redo [CTRL-Y]. As designed?

laurensvalk, 2024-02-08:
The situation above is easily fixed by changing the dropdown.

In the original screenshot the whole dropdown is missing.

BertLindeman, 2024-02-08:
Ah, I will retry.

BertLindeman, 2024-02-08:
I am using ```v3.4.0b1 (Pybricks Beta v2.4.0-beta.3)```
Might the problem be in the xboxcontroller?

If I delete the sensor or remote,  the input block keeps its device object.  
Also tested with the sensor value in an if.

```python
print(sensor_2.ambient())
```

While in your situation that disappears, (I guess). 
```python
if Button.LEFT_MINUS in .buttons.pressed()
```

So before the redo even?

Hope I do not take too much of your time....

laurensvalk, 2024-02-08:
Don't worry, thanks for helping!

I know what's broken. This is just the to-do note to fix it :smile: