pybricks/support #1438: [Bug] Shutdown not working on TechnicHub

Debenben, 2024-02-10:
**Describe the bug**
The hub is blinking blue very fast instead of shutting down. The problem appeared when updating my TechnicHub to the last Beta version:

('technichub', '3.4.0b1', 'v1.20.0-23-g6c633a8dd on 2024-02-01')

The previous version and Inventorhub do not have this problem. There are two ways to reproduce:

**First way to reproduce**

```python
from pybricks.hubs import TechnicHub
from pybricks.tools import wait
from pybricks import version

print(version)
hub = TechnicHub()
wait(100)
hub.system.shutdown()
```

**Second way to reproduce**

```python
from pybricks.hubs import TechnicHub
from pybricks.tools import wait
from pybricks import version

print(version)
hub = TechnicHub()
hub.system.set_stop_button(None)
wait(100000)
```

run the program and long-press the button while it is running.

laurensvalk, 2024-02-11:
Thanks for reporting!

We've got a next beta released scheduled for this week, but I think this issue is a high priority one to fix directly after.

laurensvalk, 2024-02-11:
This appears to be introduced in [04fd2e770d97ab4771e77072bab0d9d61bcdeb45](https://github.com/pybricks/pybricks-micropython/commit/04fd2e770d97ab4771e77072bab0d9d61bcdeb45)

BertLindeman, 2024-02-11:
Just for the record: 

The both scenario's also hit this problem on a technichub at 

```('technichub', '3.3.0', 'ci-build-3232-v3.3.0-20-ge836b04f on 2024-01-25')```

Not only the technichub though.  Seen on movehub and cityhub.

laurensvalk, 2024-02-11:
Found the likely culprit! If this fix looks okay we can include it in this week's beta.

> This appears to be introduced in https://github.com/pybricks/pybricks-micropython/commit/04fd2e770d97ab4771e77072bab0d9d61bcdeb45

While technically true (the symptoms stated there), the above commit is in fact one part of a series of proper fixes that allow us to remove an old hack that I think it a root cause of various Bluetooth-related shutdown issues.

closed 2024-02-16 by laurensvalk

BertLindeman, 2024-02-17:
Confirmed fixed on 
```
('primehub', '3.4.0b2', 'v1.20.0-23-g6c633a8dd on 2024-02-14')
('cityhub', '3.4.0b2', 'v1.20.0-23-g6c633a8dd on 2024-02-14')
```
First scenario stops the hub.
Second scenario keeps running until timeout (as programmed 100 seconds) or a long press on the hub button,
and then does the shutdown.