pybricks/support #1439: Block Coding: Logic Block Comparison "<" (less than) shows "left trigger"

ggramlich, 2024-02-11:
**Describe the bug**
The way the logic block for comparisons with option "< True if left value is smaller" is displayed, is broken since one of the last updates. It shows "left trigger" instead of "<".

**To reproduce**
Steps to reproduce the behavior:
Pull a comparison block from the Data -> Logic section into the program an switch to "<"

**Expected behavior**
The block should read "3 < 4".
It actually reads "3 left trigger 4"

**Screenshots**
[image]

laurensvalk, 2024-02-11:
Oops! Thanks for catching this.

Looks like I've used the same translation key twice. I wonder what the left trigger is for...

By the way, although we released the blocks originally on the beta channel, you can now switch it to the stable version to avoid surprises like these.

laurensvalk, 2024-02-11:
This should be fixed in the beta release update coming this week.

ggramlich, 2024-02-13:
> By the way, although we released the blocks originally on the beta channel, you can now switch it to the stable version to avoid surprises like these.

Is there a way to find out, which stable version of block coding corresponds to which beta version?
The https://github.com/pybricks/pybricks-code/blob/master/CHANGELOG.md does not seem to reflect the current state.

Do we need to deactivate the license key in beta or can we use the same license key for beta and stable?

laurensvalk, 2024-02-13:
> Is there a way to find out, which stable version of block coding corresponds to which beta version?

This part of the changelog isn't available yet, but we're working on it. Thanks!

> Do we need to deactivate the license key in beta or can we use the same license key for beta and stable?

You can use both at the same time.

[image]

laurensvalk, 2024-02-19:
Fix has been released in latest beta.

closed 2024-02-19 by laurensvalk