pybricks/support #1440: [Bug] Block Coding misses import of random when accessing random item from list.

ggramlich, 2024-02-11:
**Describe the bug**
In block coding: Trying to access a random element from list results in the error
`NameError: name 'random' isn't defined`

**To reproduce**
Setup a variable as list.
Try to get random item from list.

**Expected behavior**
It should get a random element from the list.

**Screenshots**
[image]

laurensvalk, 2024-03-07:
The app currently includes the default list blocks from Blockly. I'd like to gracefully deprecate those and add list blocks that suit the rest of the Pybricks API quite well. Programs that already have them can keep using them.

Things I'd like to avoid in the new blocks:
- Empty non-value inputs. These should be editable fields
- Shape changing in the get (value shape) and remove (statement shape)
- Block modes that generate multiple lines of code with intermediate temporary values (e.g. _set_ random value)

----

The following is probably a bit cryptic. I'm mainly posting this as a to-do list for myself. I will add screenshots of the blocks for review for discussion, later.

The main goal is to keep the most useful functionality, while making the blocks a lot simpler to use.

----

The list itself could be pre-populated with variable names in a dropdown, or kept empty like we do in the unpack block.

Suggested value shaped block:
- no mode, so only "get" operation
- Options: first, last, index, or random (so drop the option `# from end`)

Suggested statement shaped block:
- Instead of a separate set and an insert at (#, first, last, etc) as it is now, use: append, prepend, set
- this is the statement shape, so this is where the remove option belongs

laurensvalk, 2024-03-07:
For the setter, I opted for a single dropdown, instead of mixed, nested dropdowns.

So set, remove, and insert. Each with index, start, or end. We could even drop first/last, but then you'd have to remember that 0 is first and -1 is last.

------------

What do you think of this draft, @ggramlich?

[image]

ggramlich, 2024-03-07:
@laurensvalk 
This looks good to me.
I would keep the first / last options.

Could you add a screenshot of the expanded combobox of the statement block?

laurensvalk, 2024-03-07:
Thanks for looking at this! Yes, these do indeed include more detail, which is a Pybricks extension on blockly:

[image]

[image]

laurensvalk, 2024-03-07:
This is now available to try out in the latest beta version.

[image]

[image]

closed 2024-03-11 by laurensvalk