pybricks/support #1443: [Bug] Error building pybricks

afarago, 2024-02-14:
**Describe the bug**
What is the problem?
I am trying to compile pybricks-code, yet all I get is errors-errors-errors.
[output.log](https://github.com/pybricks/support/files/14273755/output.log)
Would be happy for a pointer how to proceed - trying to experiment with a few features to add :D.

**To reproduce**
Steps to reproduce the behavior:
1. cloned rep
2. followed instructions from CONTRIBUTING.md

I am trying to use Windows 11 system. Maybe that is the source of all errors.
Prior to this error needed to modify `.eslintrc.js` and `.prettierrc.js` to reach this error.

**Expected behavior**
What did you expect to happen instead?
Project builds and starts IDE.

**Screenshots**
N/A

.eslintrc.js
```
module.exports = {
    parser: '@typescript-eslint/parser',
    extends: [
        'eslint:recommended',
        'plugin:react/recommended',
        'plugin:react-hooks/recommended',
        'plugin:@typescript-eslint/eslint-recommended',
        'plugin:@typescript-eslint/recommended',
        'plugin:prettier/recommended',
        'plugin:import/errors',
        'plugin:import/warnings',
        'plugin:import/typescript',
        'prettier',
    ],
    parserOptions: {
        ecmaVersion: 2018,
        sourceType: 'module',
    },
    plugins: ['react', 'prettier'],
    rules: {
        curly: 'error',
        eqeqeq: 'error',
        'no-multi-spaces': 'error',
        'no-trailing-spaces': 'error',
        'no-multiple-empty-lines': 'error',
        '@typescript-eslint/no-unused-vars': ['error', { argsIgnorePattern: '^_' }],
        'no-unused-vars': ['error', { argsIgnorePattern: '^_' }],
        'sort-imports': ['error', { ignoreDeclarationSort: true }],
        'import/order': ['error', { alphabetize: { order: 'asc' } }],
        'react-hooks/exhaustive-deps': 'error',
        'react/jsx-no-target-blank': ['error', { allowReferrer: true }],
        'prettier/prettier': ['error', { endOfLine: 'auto' }, { usePrettierrc: true }],
    },
    settings: {
        react: { version: 'detect' },
    },
    ignorePatterns: [
        'config/**/*.js',
        'config/**/*.mjs',
        'scripts/*.js',
        'test/env.js',
        '.eslintrc.js',
        '.prettierrc.js',
        'craco.config.js',
    ],
};
```

.prettierrc.js
```
module.exports = {
    semi: true,
    trailingComma: 'all',
    singleQuote: true,
    printWidth: 88,
    tabWidth: 4,
    endOfLine: 'auto',
};
```

dlech, 2024-02-14:
Does it work if you revert the changes to the config files? Are you sure you are using the right version of NodeJS? Does it work if you delete the `node_modules` folder then run `yarn` again?

afarago, 2024-02-14:
For the time being Windows has CR problems with the original config files.
Just tried it on another machine, clean install, same result.

As a next step I will be retesting in a few days with WSL2 to mimic linux/Mac environment.
BTW: would be great to check -- are any constraints for the build environment? TIA!

dlech, 2024-02-20:
FYI, I just added a devcontainer config file to help set up a known working environment.

afarago, 2024-02-25:
Thank you, that is definitely helpful.
I have tried it with WSL2 on a WIndows machine - same error came and build failed. It might be my lack of knowledge yet, I will try as I have some ideas to PoC.

If there are any hints for building this in a Win machine even with WSL or docker, that is awesome.

dlech, 2024-02-25:
If you have Docker installed on Windows and open the project in VS Code with the `ms-vscode-remote.remote-containers` extension installed, it should take care of everything automatically after that except for `yarn start`.

When you tried in WSL, was the source code and node modules installed in a case-sentitive file system?

afarago, 2024-07-10:
I was able to come over this issue finally!
For others trying to build pybricks-code on a windows machine.

WSL2 + Ubuntu
do not use any windows directory (/mnt/c), rather checkout under /home/....
* windows based git if would be by default CRLF based
* git local checkout is much faster for building

closed 2024-07-10 by afarago