pybricks/support #1453: [Feature] Move Hub build size

laurensvalk, 2024-02-18:
**Is your feature request related to a problem? Please describe.**
It's that time of year again where the Move Hub is almost out of space.

This makes it difficult to merge small fixes, e.g. to motor drivers.

I think the Move Hub is pretty much feature complete, so we shouldn't need all that much.

**Describe the solution you'd like**

One of these might work:
- Optimize other code to find space.
- Disable some functionality.
- Reduce user code space (now 4K, which is higher than the 2K we had originally).

laurensvalk, 2024-02-18:
For context, it's not currently as high as it has been several times previously; we just increased the user space the last time we found a lot of space.

[image]

dlech, 2024-02-18:
There have been quite a few size reductions in upstream MicroPython in the last few releases, so I think the first step should be to update that.

laurensvalk, 2024-02-19:
Great idea!

laurensvalk, 2024-03-01:
This should keep Move Hub going for another year or more: https://github.com/pybricks/pybricks-micropython/pull/240

Especially if we only use that space for bug fixes rather than new features.

laurensvalk, 2024-03-05:
With about 2.8Kb free for future bug fixes even after adding a substantial new feature (the `Car` class), I think this is good enough for this issue.

closed 2024-03-05 by laurensvalk