pybricks/support #1457: Issues with installed pybricksdev on Mac 14.3.1 (ARM)

bcfletcher, 2024-02-19:
i have mpy-cross 1.22 installed.

pip keeps wanting to install old version versions of pybricksdev not rev 46 

i'm running

pip3 install pybricksdev
Collecting pybricksdev
  Downloading pybricksdev-1.0.0a17-py3-none-any.whl (196 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 196.4/196.4 kB 2.7 MB/s eta 0:00:00
Collecting aioserial<2.0.0,>=1.3.0 (from pybricksdev)
  Downloading aioserial-1.3.1.tar.gz (9.6 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting appdirs<2.0.0,>=1.4.4 (from pybricksdev)
  Downloading appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting argcomplete<2.0.0,>=1.11.1 (from pybricksdev)
  Downloading argcomplete-1.12.3-py2.py3-none-any.whl (38 kB)
Collecting asyncssh<3.0.0,>=2.2.1 (from pybricksdev)
  Downloading asyncssh-2.14.2-py3-none-any.whl.metadata (9.9 kB)
Collecting bleak<0.13.0,>=0.12.1 (from pybricksdev)
  Downloading bleak-0.12.1-py2.py3-none-any.whl (123 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 123.1/123.1 kB 10.7 MB/s eta 0:00:00
INFO: pip is looking at multiple versions of pybricksdev to determine which version is compatible with other requirements. This could take a while.
Collecting pybricksdev
  Downloading pybricksdev-1.0.0a16-py3-none-any.whl (195 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 195.2/195.2 kB 7.7 MB/s eta 0:00:00
  Downloading pybricksdev-1.0.0a15-py3-none-any.whl (197 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 197.6/197.6 kB 9.5 MB/s eta 0:00:00
  Downloading pybricksdev-1.0.0a14-py3-none-any.whl (197 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 197.5/197.5 kB 14.1 MB/s eta 0:00:00
  Downloading pybricksdev-1.0.0a13-py3-none-any.whl (196 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 196.6/196.6 kB 12.8 MB/s eta 0:00:00
  Downloading pybricksdev-1.0.0a12-py3-none-any.whl (196 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 196.4/196.4 kB 10.2 MB/s eta 0:00:00
  Downloading pybricksdev-1.0.0a11-py3-none-any.whl (195 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 196.0/196.0 kB 9.5 MB/s eta 0:00:00
Collecting bleak==0.12.0 (from pybricksdev)
  Downloading bleak-0.12.0-py2.py3-none-any.whl (123 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 123.2/123.2 kB 10.4 MB/s eta 0:00:00
Collecting pybricksdev
  Downloading pybricksdev-1.0.0a10-py3-none-any.whl (195 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 195.3/195.3 kB 11.5 MB/s eta 0:00:00
INFO: pip is still looking at multiple versions of pybricksdev to determine which version is compatible with other requirements. This could take a while.
ERROR: Cannot install pybricksdev==1.0.0a10, pybricksdev==1.0.0a11, pybricksdev==1.0.0a12, pybricksdev==1.0.0a13, pybricksdev==1.0.0a14, pybricksdev==1.0.0a15, pybricksdev==1.0.0a16 and pybricksdev==1.0.0a17 because these package versions have conflicting dependencies.

The conflict is caused by:
    pybricksdev 1.0.0a17 depends on mpy-cross==1.14
    pybricksdev 1.0.0a16 depends on mpy-cross==1.14
    pybricksdev 1.0.0a15 depends on mpy-cross==1.14
    pybricksdev 1.0.0a14 depends on mpy-cross==1.14
    pybricksdev 1.0.0a13 depends on mpy-cross==1.14
    pybricksdev 1.0.0a12 depends on mpy-cross==1.14
    pybricksdev 1.0.0a11 depends on mpy-cross==1.14
    pybricksdev 1.0.0a10 depends on mpy-cross==1.14

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

PIP STDOUT
----------
Collecting pybricksdev
  Using cached pybricksdev-1.0.0a17-py3-none-any.whl (196 kB)
Collecting aioserial<2.0.0,>=1.3.0 (from pybricksdev)
  Using cached aioserial-1.3.1.tar.gz (9.6 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting appdirs<2.0.0,>=1.4.4 (from pybricksdev)
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting argcomplete<2.0.0,>=1.11.1 (from pybricksdev)
  Using cached argcomplete-1.12.3-py2.py3-none-any.whl (38 kB)
Collecting asyncssh<3.0.0,>=2.2.1 (from pybricksdev)
  Using cached asyncssh-2.14.2-py3-none-any.whl.metadata (9.9 kB)
Collecting bleak<0.13.0,>=0.12.1 (from pybricksdev)
  Using cached bleak-0.12.1-py2.py3-none-any.whl (123 kB)
INFO: pip is looking at multiple versions of pybricksdev to determine which version is compatible with other requirements. This could take a while.
Collecting pybricksdev
  Using cached pybricksdev-1.0.0a16-py3-none-any.whl (195 kB)
  Using cached pybricksdev-1.0.0a15-py3-none-any.whl (197 kB)
  Using cached pybricksdev-1.0.0a14-py3-none-any.whl (197 kB)
  Using cached pybricksdev-1.0.0a13-py3-none-any.whl (196 kB)
  Using cached pybricksdev-1.0.0a12-py3-none-any.whl (196 kB)
  Using cached pybricksdev-1.0.0a11-py3-none-any.whl (195 kB)
Collecting bleak==0.12.0 (from pybricksdev)
  Using cached bleak-0.12.0-py2.py3-none-any.whl (123 kB)
Collecting pybricksdev
  Using cached pybricksdev-1.0.0a10-py3-none-any.whl (195 kB)
INFO: pip is still looking at multiple versions of pybricksdev to determine which version is compatible with other requirements. This could take a while.

The conflict is caused by:
    pybricksdev 1.0.0a17 depends on mpy-cross==1.14
    pybricksdev 1.0.0a16 depends on mpy-cross==1.14
    pybricksdev 1.0.0a15 depends on mpy-cross==1.14
    pybricksdev 1.0.0a14 depends on mpy-cross==1.14
    pybricksdev 1.0.0a13 depends on mpy-cross==1.14
    pybricksdev 1.0.0a12 depends on mpy-cross==1.14
    pybricksdev 1.0.0a11 depends on mpy-cross==1.14
    pybricksdev 1.0.0a10 depends on mpy-cross==1.14

To fix this you could try to:
1. loosen the range of package versions you've specified

bcfletcher, 2024-02-19:
so i tried downloading the a64 .whl file and installing it but it doesn't appear to recognize python 3.12.2

user@bigsky Python % pip3 install pybricksdev-1.0.0a46-py3-none-any.whl 
Processing ./pybricksdev-1.0.0a46-py3-none-any.whl
Collecting aioserial<2.0.0,>=1.3.0 (from pybricksdev==1.0.0a46)
  Using cached aioserial-1.3.1.tar.gz (9.6 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting appdirs<2.0.0,>=1.4.4 (from pybricksdev==1.0.0a46)
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting argcomplete<2.0.0,>=1.11.1 (from pybricksdev==1.0.0a46)
  Using cached argcomplete-1.12.3-py2.py3-none-any.whl (38 kB)
Collecting asyncssh<3.0.0,>=2.2.1 (from pybricksdev==1.0.0a46)
  Using cached asyncssh-2.14.2-py3-none-any.whl.metadata (9.9 kB)
Collecting bleak<0.21.0,>=0.20.2 (from pybricksdev==1.0.0a46)
  Downloading bleak-0.20.2-py3-none-any.whl.metadata (4.5 kB)
Collecting hidapi<0.14.0,>=0.13.1 (from pybricksdev==1.0.0a46)
  Downloading hidapi-0.13.1.tar.gz (130 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 130.2/130.2 kB 2.1 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting mpy-cross-v5<2.0.0,>=1.0.0 (from pybricksdev==1.0.0a46)
  Downloading mpy_cross_v5-1.0.1-py3-none-macosx_11_0_arm64.whl.metadata (591 bytes)
Collecting mpy-cross-v6<2.0.0,>=1.0.0 (from pybricksdev==1.0.0a46)
  Downloading mpy_cross_v6-1.0.1-py3-none-macosx_11_0_arm64.whl.metadata (591 bytes)
Collecting packaging<23,>=22 (from pybricksdev==1.0.0a46)
  Downloading packaging-22.0-py3-none-any.whl (42 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.6/42.6 kB 3.8 MB/s eta 0:00:00
Collecting prompt-toolkit<4.0.0,>=3.0.18 (from pybricksdev==1.0.0a46)
  Downloading prompt_toolkit-3.0.43-py3-none-any.whl.metadata (6.5 kB)
Collecting pyusb<2.0.0,>=1.0.2 (from pybricksdev==1.0.0a46)
  Downloading pyusb-1.2.1-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.4/58.4 kB 5.5 MB/s eta 0:00:00
Collecting reactivex<5.0.0,>=4.0.4 (from pybricksdev==1.0.0a46)
  Downloading reactivex-4.0.4-py3-none-any.whl (217 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 217.8/217.8 kB 8.8 MB/s eta 0:00:00
Collecting semver<3.0.0,>=2.13.0 (from pybricksdev==1.0.0a46)
  Downloading semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting tqdm<5.0.0,>=4.62.3 (from pybricksdev==1.0.0a46)
  Downloading tqdm-4.66.2-py3-none-any.whl.metadata (57 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 57.6/57.6 kB 5.0 MB/s eta 0:00:00
Collecting typing-extensions<5.0.0,>=4.3.0 (from pybricksdev==1.0.0a46)
  Downloading typing_extensions-4.9.0-py3-none-any.whl.metadata (3.0 kB)
Collecting validators<0.19.0,>=0.18.2 (from pybricksdev==1.0.0a46)
  Downloading validators-0.18.2-py3-none-any.whl (19 kB)
INFO: pip is looking at multiple versions of pybricksdev to determine which version is compatible with other requirements. This could take a while.
ERROR: Package 'pybricksdev' requires a different Python: 3.12.2 not in '<3.12,>=3.8.1'

user@bigsky Python % python3 --version
Python 3.12.2

dlech, 2024-02-19:
> ERROR: Package 'pybricksdev' requires a different Python: 3.12.2 not in '<3.12,>=3.8.1'

It is not currently compatible with Python 3.12. You will need to use 3.11.