pybricks/support #1462: [Bug] Technic Hub has to disconnect from some host PCs to connect to Xbox Controller

laurensvalk, 2024-02-20:
**Describe the bug**
When the Technic Hub is connected to Pybricks Code on some host computers, the hub is able to find and connect to the Xbox controller, but not read any characteristics or pair.

On some computers and phones, it works just fine. It works perfectly on my PC and laptop and phone, but not on the PCs of most people who have tried it. Things seem to be pretty good on most Android phones we've tried.

So to guarantee that the Xbox Controller "works" for everyone, we decided to automatically disconnect from the PC, just to be sure. But this should be considered a workaround. It would be better to fix the actual problem.

The workaround is less than ideal since you can't see any program output or errors. You also have to reconnect every time you want to change the program. And walk over to the hub first to stop the current program.

**To reproduce**

Run this program. Notice the `stay_connected=True` parameter, which avoids our current workaround of auto-disconnecting.

```python
from pybricks.iodevices import XboxController
from pybricks.tools import wait
from pybricks.hubs import TechnicHub
from pybricks.parameters import Color

hub = TechnicHub()
hub.light.on(Color.GREEN)

# To guarantee it "works" for everyone, we automatically disconnect from the PC.
# But it would be much nicer to stay connected. Who wants to try it with the
# following setting? We want to figure out why this works fine on some computers
# and some phones, but not at all on others.
xbox = XboxController(stay_connected=True)

print("Connected!")

state = None

# This loop just changes the color light when you press a button, to see that
# things are working. Nothing special here.
while True:
    new = xbox.buttons.pressed()
    if new == state:
        continue
    state = new
    hub.light.on(Color.VIOLET if state else Color.RED)
    print(state)
```

Tagging @nstrijbosch and @Tcm0 for feedback. Does it work on your computers?

**Note**
Assuming we can fix it eventually, it would be great if we can keep the discussion about the `stay_connected` parameter here on GitHub. It's an unofficial parameter that will be removed when we fix this issue. Let's try not to over-complicate tutorials  :smile:

Tcm0, 2024-02-20:
It doesn't work with my PC. It either times out or has that GATT issue.
My bluetooth driver is from intel. Maybe it's connected to that because intel is probably more common with desktop pcs?

laurensvalk, 2024-02-20:
> It doesn't work with my PC. (...) It times out.

Thanks! This is indeed the "expected" result when the hub fails to communicate with the controller for some reason. Maybe we'll get back to you at some point to run some tests :)

> has that GATT issue.

I have seen this on some people's Windows computers recently, and I think it's unrelated to the Xbox stuff. It also sometimes happens with other programs.

If you see this again, would you mind sharing a screenshot or logs, perhaps in a new issue? It would be good to address that. Thanks!

laurensvalk, 2024-02-20:
My hunch* is that since the host PC sets the connection parameters with the hub, it may conflict with the controller's parameters with the hub. I wonder if we could set different values from the host side. On the other hand, the hub requests the same link parameters for every host, but I don't know if the host actually accepts them?

(*I'm thinking this because, in a somewhat similar fashion, broadcasting did not work if we accepted the controller's faster defaults. We reject those now, which makes broadcasting work. Maybe if certain timing intervals overlap exactly it won't do certain operations.)

BertLindeman, 2024-02-21:
The TechnicHub is not connecting if the argument `stay_connected=True`  is added on Win11 23H2 and Edge.
Timeout . . . 

This TechnicHub is:
```
('technichub', '3.4.0b2', 'v1.20.0-23-g6c633a8dd on 2024-02-14')

This is hub "hubB-v3.4.0b2" as a technichub
        Bluetooth chip version "v3.02.00"
        battery voltage:        7.480 V
```

laurensvalk, 2024-02-26:
This is starting to sound a lot like https://github.com/pybricks/support/issues/397. @dlech, do you remember if in #397 it timed out on the scan or timed out on the first operation after connecting, as we have here?

gyenesvi, 2024-02-26:
On my really old Samsung phone, the website `beta.pybricks.com` does not seem to load (blank screen). That's all I have for Android..

laurensvalk, 2024-02-26:
> (On the phone it's not doable.)

Since the code is already on the hub, you'd just have to connect via the Bluetooth button and press the green button to start it :)

Thanks for confirming that it does work on your Android. That's a 100% score so far (but just about 10% for PC/Mac)...

BertLindeman, 2024-02-26:
Also OK on android 11.
A bit smaller program to type less....
[image]

laurensvalk, 2024-02-26:
Thanks!

I meant no typing was needed at all if the code was already on the hub, but thanks anyway! 😄

Now we just have to find out why it works on Android but not on most computers.

BertLindeman, 2024-02-27:
> I meant no typing was needed at all if the code was already on the hub, but thanks anyway! 😄

I was not on sharpest yesterday. Of course: one button press . . . 
Can confirm that on android 4 beta.pybricks.com does not work. 😉 
Let me know if I can do additional testing .

Tcm0, 2024-02-27:
> @Tcm0 , @BertLindeman , @gyenesvi , would you mind trying it on your Android phone also?
> 
> Thanks!

Android tablet. Android version 8.1.0.
The hub doesn't find the controller when connected to the tablet. Pressing the stop button in the software results in an unknown GATT error. The connection worked when I set that stayConnected parameter to false.