pybricks/support #1467: [Bug] ImportError: can't import name XboxController

ninoguba, 2024-02-22:
**Describe the bug**
Getting this error in the output when running the project code for the Audi + Xbox.

ImportError: can't import name XboxController

**To reproduce**
Steps to reproduce the behavior:
1. Follow steps to install Pybricks firmware onto Technic hub
2. Connect onto the Technic hub
3. Create new file > python
4. Copy the sample project
5. Run the program

**Expected behavior**
No errors

I've tried this on two different computers - one is a Mac the other a Windows - and two different Technic hubs.  Same thing happens. =(

ninoguba, 2024-02-22:
Ok I've now tried this with the beta firmware and in the middle of running the program, before it finished downloading to the hub, I got an unexpected error message popup stating "GATT Error Unknown."

ninoguba, 2024-02-22:
I've tried a second time running the same program and to my surprise it worked!   So I'm now able to connect the Xbox controller to the hub and finally control the model!  Yey!

closed 2024-02-22 by ninoguba

laurensvalk, 2024-02-22:
Just adding a few details for anyone else running into this.

> ImportError: can't import name XboxController

An `ImportError` means that a particular class is not available. As you've found, it is indeed available in the beta version.

To try it, click on "Try New Beta Features" under settings.

[image]

In a few weeks, this should also be available in the stable version, so using the beta won't be needed anymore.