pybricks/support #1468: [Bug] Popup stating "GATT Error Unknown."

laurensvalk, 2024-02-22:
(...) with the beta firmware and in the middle of running the program, before it finished downloading to the hub, I got an unexpected error message popup stating "GATT Error Unknown."

_Originally posted by @ninoguba in https://github.com/pybricks/support/issues/1467#issuecomment-1958659447_

laurensvalk, 2024-02-22:
This was also reported by:

_Originally posted by @Tcm0 in https://github.com/pybricks/support/issues/1462#issuecomment-1954820503_

laurensvalk, 2024-02-22:
I have also seen this while working on @martynboogaarts 's laptop.

Once it did occur, it was deterministic after that. We kept getting that error when we clicked Download again.

Restarting the hub worked around it.

Tcm0, 2024-02-22:
So far I only got the issue once and it didn't happen when I tested my code for the app controlled transformation vehicle (I did at least 10 program uploads for sure). It only happened with that Technic Hub Disconnection test code

PeterAmiga, 2024-04-01:
I am trying to install firmware on boost hub, and I get above error or attached error 'something went wrong'. I have tried ten times. I tried to install official firmware twice, and that worked fine.
I follow the instructions, setting no specific hub name. Instructions say Select Lego Bootloader and click Connect. But I can see Lego Bootloader - Paired, and then I can press Pair.

For me this does not seem logic, since it is already paired.

The process starts but I get attached error after around 10-12 seconds and the flashing on the hub stops.

Is there any way for me to see additional logging about what could go wrong?
[image]
[image]

PeterAmiga, 2024-04-01:
I just fetched bluetooth logging from browser. Will try other debugging tomorrow.
[image]

dlech, 2024-04-01:
For this particular error ("The hub said something went wrong"), I think we will need Bluetooth packet logs.

The console log in the browser may give a slightly better hint as to the cause though (narrows it down to 1 of 5 possibilities).

BertLindeman, 2024-04-01:
Comment of a simple Pybricks user :

1. The "pair" in pybricks in fact means to connect to.
2. I had this type of error sometimes when there were too many devices near the pc and the hub.
As example when my phone is between the pc and the hub.

But bluetooth logging might show what is the real problem.

Bert

PeterAmiga, 2024-04-02:
Impressively quick replies :-)

I realised I could run pybricks from my android phone. I also used my android phone to install the original firmware again. I succeded installing pybricks firmware using my phone.

And afterwards I can run a program on the hub from my Windows pc.

Thanks for the suggestions on how to solve the pc problem. For now I will experiment with the code on my hub and not dig further into this problem.

If you would like to have the Bluetooth packet logs, let me know. I will try to provide them.

dlech, 2024-04-02:
> If you would like to have the Bluetooth packet logs, let me know. I will try to provide them.

Yes, that wold be helpful to be able to see if this is a bug we can fix or if it is just a bad Bluetooth driver on Windows.

kennethdammyr, 2025-10-25:
Got the same error ````GATT Error Unknown.```` when trying to download code to a Technic Hub. Some downloads work, others don't. There shouldn't be any interference between the laptop and hub and the signal is strong.
I'm not sure if it is related, but at the same time I am having trouble getting my full code to run. It breaks when a Xbox-controller is connected. Could it be that this errors depends on the code or is it strictly a connectivity-thing?