pybricks/support #1470: Tags for releases v3.4.0b1 and v3.4.0b2 are missing

dlech, 2024-02-22:


laurensvalk, 2024-02-22:
It looks like the Move Hub CI build for releases is slightly larger than master, and larger still than local builds, so the CI build for the tag fails by 4 bytes.

laurensvalk, 2024-02-27:
I think we can do the tags and just let the move hub build fail.

v3.4.0b3 should be out soon, properly released via CI after https://github.com/pybricks/pybricks-micropython/pull/239 or something like it gets merged.

laurensvalk, 2024-02-28:
Tags are here. We also saved 200 bytes just now, so the `v3.4.0b3` should release normally.

Everything else on the master branch did build on CI by the way, it's just the tagged build that was apparently just slightly too big.

laurensvalk, 2024-03-05:
Also seems to miss update to Ubuntu 22.04. Dependency on mpy cross and handling of git hashes also seems to be handled differently.

Is this something you might have time for @dlech ?

laurensvalk, 2024-03-05:
I guess this will do:

```diff
diff --git a/.github/workflows/release.yml b/.github/workflows/release.yml
index 00404d9f7..3201db9f1 100644
--- a/.github/workflows/release.yml
+++ b/.github/workflows/release.yml
@@ -11,26 +11,15 @@ env:
 jobs:
   upload_release:
     name: Upload Release Assets
-    runs-on: ubuntu-20.04
+    runs-on: ubuntu-22.04
     steps:
-      - name: Check cache
-        id: check-cache
-        uses: actions/cache@v2
-        with:
-          path: ${{ runner.tool_cache }}/gcc-arm-10-2020-q4
-          key: gcc-arm-10-2020-q4
-      - name: Download
-        if: steps.check-cache.outputs.cache-hit != 'true'
-        uses: dlech/arm-none-eabi-gcc@master
-        with:
-          release: '10-2020-q4'
-          directory: ${{ runner.tool_cache }}/gcc-arm-10-2020-q4
-      - name: Add cross-compiler to path
-        run: echo ${{ runner.tool_cache }}/gcc-arm-10-2020-q4/bin >> $GITHUB_PATH
+      - name: Install cross-compiler
+        run: sudo apt-get update && sudo apt-get install --yes gcc-arm-none-eabi
       - name: Checkout code
         uses: actions/checkout@v3
         with:
           submodules: true
+          fetch-depth: 0
       - name: Build firmware
         run: |
           make $MAKEOPTS -C micropython/mpy-cross

```

laurensvalk, 2024-03-05:
Maybe the create release is outdated? https://stackoverflow.com/questions/75679683/how-can-i-auto-generate-a-release-note-and-create-a-release-using-github-actions

laurensvalk, 2024-03-06:
Fixed by https://github.com/pybricks/pybricks-micropython/commit/31787a660703c3a8c8ffdd490f0c4a9eec698e46.

`gh release create` can also upload assets, so we can do it all in one step.

We have to rename the zip files locally because it gave `404` errors if they were all `firmware.zip` despite a unique `LABEL`.

closed 2024-03-06 by laurensvalk