pybricks/support #1472: [Bug] hub.display.animate doesn't work with multitask

MrWonka, 2024-02-23:
**Describe the bug**
hub.display.animate doesn't work with multitask

**To reproduce**
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, ColorDistanceSensor, ForceSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.tools import wait, StopWatch, run_task, multitask, Matrix

hub = InventorHub()
button = ForceSensor(Port.A)

press2 = Matrix(
    [
        [0, 75, 0, 0, 0],
        [0, 75, 0, 0, 0],
        [0, 0, 0, 0, 0],
        [0, 100, 100, 100, 0],
        [100, 100, 100, 100, 100],
    ]
)
press3 = Matrix(
    [
        [0, 75, 0, 0, 0],
        [0, 0, 0, 0, 0],
        [0, 0, 0, 0, 0],
        [0, 100, 100, 100, 0],
        [100, 100, 100, 100, 100],
    ]
)

async def buttonDisplay():
    temp = 0
    while True:
        while await button.pressed():
            hub.display.off()
            await wait(5000)
        hub.display.animate([press3,press2],500)

async def runGame():
    await multitask(buttonDisplay())

run_task(runGame())

BertLindeman, 2024-02-23:
I do not have the forceSensor so I commented that one out and the button.pressed while.
And I added a wait to it.

So that function looks like:
```python
async def buttonDisplay():
    temp = 0
    while True:
        # while await button.pressed():
        #     hub.display.off()
        #     await wait(5000)
        hub.display.animate([press3,press2],500)
        await wait(1000)
```

Now the display neatly blinks.
Might it be that there is nothing else to do, so the program stops?

(No asynchronous experience here)

Bert

MrWonka, 2024-02-23:
Oh yep. That fixes it. Thank you!

closed 2024-02-23 by MrWonka