pybricks/support #1474: [Feature] Car class possible improvements?

gyenesvi, 2024-02-24:
I see that the latest beta version has a Car class for easy control of motors that make up drive/steering of cars. This is a nice initiative, and I have been wondering about a few details about such a class, so I'd like to understand some decisions and make some proposals. I am assuming that the purpose of this class is to make control/programming easier, to hide some of the details of mapping control signals (for example coming from the xbox controller) to motor commands, and also calibration stuff. So in a sense this class is probably not adding fundamentally new functionality that is not already possible with existing classes such as the motors and additional Python level glue code. Correct me if that assumption is wrong.

One thing is that I would assume that such a class could get pretty complex in the future as there are many possibilities for building cars in terms of drive systems, steering systems and add-ons such as gearboxes, winches, etc., so the class might gain a lot of parameters/config options and methods in the future. So I was wondering it it would be a better idea to model parts of it separately (individual classes), such a steering system, a drive system, a gearbox system, etc. This way these individual parts would get less complex, and they could even have different implementations for various purposes (for example regular drive vs tracked drive, single axle steering vs multi axle steering, etc), that could be combined in many ways as necessary.

The other thing I wonder is whether such classes actually need to be the part of the core operating system, and whether they need to be programmed in C++ with a python wrapper (that's what I would as I looked into the code). If my assumption above is correct, then it could probably suffice to do this part in pure Python, which could have a couple advantages; the code could be easier to examine, modify, build on (for example by inheriting from and overriding just a few methods). Maybe it would be possible to import only the bits that are actually required, not bloating total memory usage (not sure if that's true, depending how the whole op system works), and that could allow more parameters/options to be added without being afraid of bloating the core operating system code.

This latter one is an interesting question in general, whether the Pybricks system allows for pure Python classes to be added as utility libraries building on core system functionality without a memory usage penalty. That could allow a nice ecosystem of reusable utility classes. For example, I also have ideas for XboxController utilities.

laurensvalk, 2024-02-24:
> I see that the latest beta version has a Car class for easy control of motors that make up drive/steering of cars (...) the purpose of this class is to make control/programming easier (...) So in a sense this class is probably not adding fundamentally new functionality (...) Correct me if that assumption is wrong.

Completely correct! See https://github.com/pybricks/support/issues/1397

>  If my assumption above is correct, then it could probably suffice to do this part in pure Python

That is indeed how we've done it. We've not done this before because the Move Hub did not have enough space for it. It isn't going to get new features now anyway since it is entirely full, so new features are allowed to take a bit more space.

The `pybricks.robotics` module is a good place for higher level abstractions around motors. The `DriveBase` has been there since the beginning, and now we've added the `Car`. We might also have a `MultiMotorSync`, and so on.

We might keep the `Car` very simple, but we could certainly have other classes as well.

If it gets really specific, containing features that most people might not use, then it may be better suited for an example project instead of shipping it with the firmware.

gyenesvi, 2024-02-24:
> That is indeed how we've done it.

Okay, that's great to hear. I did not manage to find that Python code before, only found some C++ interface code for the Car class, that's why I assumed it's not in Python.

So I entirely agree that such utility classes don't need to be shipped with the core code, but could reside in some user library that the main program can include. I was wondering though whether such libraries could be shared in a way that are easily importable for users of code.pybricks.com, without needing to copy. For example if I make separate python source files on code.pybricks.com, I can import them into my main program, but if I wanted to share them with others I'd have to put them out somewhere, they'd have to copy them into separate files, etc. I was wondering if there could be a place to upload libraries (maybe in a reviewed manner) that could become importable for any user of code.pybricks.com?

For example, I have a class now that deals with drive separately, and has a method for calibrating the top speed automatically to the actual model at hand. In my experiments, I have found that the rated top speed of the motors can undershoot even for realistic builds. I am using a 1kg+ medium sized model driven by 2L motors, but with proper 9V batteries instead of 7.2V rechargeables, and in this case the top speed can reach 1650 deg/s, but is capped at 1500 by the rated top speed. So my class resets the allowed top speed to a parameterized value (say 2000), measures the actual top speed, and uses that in actual commands. Btw, I can achieve really smooth control finally, at very low speeds :)

laurensvalk, 2024-03-05:
Should we keep this issue open? If not, we can close it, and open issues for one or more specific improvements when they are needed.

gyenesvi, 2024-03-06:
> We might keep the Car very simple, but we could certainly have other classes as well.

So I understand that you want to keep the car class simple but would consider other classes at some point. What's not clear to me is where those classes would reside if they are not part of the shipped operating system, and how they could become sharable in the community in an easy to reuse way. My idea would be to share some classes that implement often used individual but possibly more complex components such as setup/calibration of steering, drive motors, gearboxes, saving/loading of calibration data, helper utilities for handling button events for the remote/xbox controller, etc, and I'm sure there would be other examples as well; essentially reusable higher level code. Having such components accessible would allow me to write actual control profiles in a cleaner / simpler way, and sharing those would be the ultimate goal. What I'd like to avoid is sharing code that has dependencies on other pieces of code that needs to be located somehow and copied into the browser in a way that's importable..

So this may become a discussion with a different focus, but still it would be good to have it somewhere and not let it fade into oblivion. Shall I open a new issue with the above summary and a different title?

laurensvalk, 2024-03-06:
> So I understand that you want to keep the car class simple but would consider other classes at some point.

Yes.

> What's not clear to me is where those classes would reside

Let's consider that when there is something to add. We can open an issue for each one.

> So this may become a discussion with a different focus, but still it would be good to have it somewhere and not let it fade into oblivion. Shall I open a new issue with the above summary and a different title?

Sure! I have something in mind for easy profiles, but this is likely for a release further into the future.

closed 2024-03-06 by laurensvalk