pybricks/support #1481: [Bug] Error when running `make technichub`

Mantarri, 2024-02-25:
**Describe the bug**

As it says on the tin, `make` fails to build, at least for `technichub`. More detailed info in "screenshot."

Environment:

- **OS:** openSUSE Tumbleweed (last updated Feb. 24, 2024 as its a rolling-release) running in VirtualBox 7.0.14
- **Allocated RAM:** 6GB
- **Allocated CPU Cores:** 2
- **Host CPU:** Ryzen 5 5600X
- **Host RAM:** 32GB
- **Python:** 3.10.13
- **Poetry:** 1.7.1
- **arm-gcc:** 10-2020-q4-major
- **Make:** 4.4.1
- **CMake:** 3.28.1

I followed the setup guide in `contributing.md` (though I did get this potential error at the end of `poetry install`: `The current project could not be installed: No file/folder found for package pybricks-micropython`)

**To reproduce**
Steps to reproduce the behavior:
1. Open terminal in `pybricks-micropython` folder
2. run `make technichub`

**Expected behavior**
`make` to succeed.

**Screenshots** (a screenshot seemed less useful. I hope this isn't too long.)
```
make[1]: Entering directory '/home/mantadev/Documents/git/lego/pybricks/pybricks-micropython/micropython/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC ../py/stackctrl.c
../py/stackctrl.c: In function ‘mp_stack_ctrl_init’:
../py/stackctrl.c:32:32: error: storing the address of local variable ‘stack_dummy’ in ‘mp_state_ctx.thread.stack_top’ [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
../py/stackctrl.c:31:18: note: ‘stack_dummy’ declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from ../py/runtime.h:29,
                 from ../py/stackctrl.c:27:
../py/mpstate.h:296:23: note: ‘mp_state_ctx’ declared here
  296 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make[1]: *** [../py/mkrules.mk:83: build/py/stackctrl.o] Error 1
make[1]: Leaving directory '/home/mantadev/Documents/git/lego/pybricks/pybricks-micropython/micropython/mpy-cross'
make: *** [Makefile:123: mpy-cross] Error 2
```
Note: I also posted an issue on the `pybricks-micropython` repository, but upon looking further, it seems this might be the preferred place for issues?

laurensvalk, 2024-02-26:
Which `pybricks-micropython` commit have you checked out?

Also if you go to

```
cd pybricks-micropython/micropython/
```

```
git log
```

Then which commit of `micropython` is currently checked out?

Mantarri, 2024-02-26:
`pybricks-micropython` is `32e3cb036d21d47d77e2f9bca6294bdc22a87c28`

`micropython` submodule is `6c633a8dd9f7dd933dae709e9c697330a5d56cca`

Mantarri, 2024-02-27:
When looking around, I noticed that the `micropython` submodule `pybricks-micropython` is pulling from is rather old, and that `micropython` upstream had this issue and it was fixed. Is the `micropython` fork that `pybricks-micropython` is pulling from supposed to be last committed to 10 months ago? Did it just not get this fix?

laurensvalk, 2024-02-27:
These look like the correct commits. It is also building fine on the CI. In case it makes a difference, we use:

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
```

Yes, we occasionally rebase MicroPython to the latest version, about once a year. There is an open ticket to pull in the latest one: https://github.com/pybricks/support/issues/1459

laurensvalk, 2024-02-27:
Good to know that it's working!

closed 2024-02-27 by laurensvalk

Mantarri, 2024-02-27:
After retrying, it must have been due to me previously trying a build with the cmake option to ignore the warning. Following the dev environment setup instructions on a fresh clone ended up in the same error, even with the newer `arm-gcc` version (which actually did exactly match the version you said you're using - there are two parts of the version tag.) I decided to try spinning up a vanilla Ubuntu 22.04 VM and I didn't experience any build problems there, so it must be something with the version of one-or-more of the system dependencies (i.e. `gcc, make`) or a difference in configuration for one-or-more packages between openSUSE Tumbleweed and Ubuntu.

Will also see about trying with my popOS (non-vm) installation, as well as the arch one on my lapptop to further pin things down at some point. As it stands, all I can say is openSUSE Tumbleweed didn't work, and Ubuntu did.