pybricks/support #1497: [Bug] Spike Prime hub is stuck with spinning animation and rapid blinking button

SuneLindhard, 2024-02-29:
**Describe the bug**
Hub gets stuck with spinner and rapid blinking button.

I have a bunch of LEGO Spike Prime robots running a lot of hours a day. A few times a week, one of them will get into this state, which is difficult to get out of. I can't reset the hub by holding down the button as normally. I have to break apart the build and remove and insert the battery again.

They are running Pybricks 3.2.0, so I have considered trying to upgrade them, but this is not super easy in my setup.

I tried searching for documentation on error codes or similar but didn't find the meaning of the spinner and blinking.

I can't connect REPL to the Hub when it gets in this state.

**To reproduce**
Unfortunately I can't really describe how to reproduce, so maybe it's more a question if anyone has seen this behaviour and knows what it means. 

**Expected behavior**
Hub continues executing and doesn't freeze

**Screenshots**

https://github.com/pybricks/support/assets/11087919/aaad83fd-ffd9-46fb-8d65-45f3f5625e60

laurensvalk, 2024-02-29:
Thanks for reporting! This sounds like it could be an issue that we have recently fixed.

Basically, this situation could occur when the hub tries to cancel or shutdown _while_ a program is running. The running program failed to stop properly, so the hub keeps on waiting and blinking the shutdown animation, but never quite gets there.

One of the situations where this happens is automatic shutdown on low battery, which sounds just like the thing that would occasionally happen during busy workshops like yours.

If you like, you could try it with the latest version at https://beta.pybricks.com/

We hope to bring these updates to the stable channel https://code.pybricks.com/ in a week or two.

sultrifork, 2024-03-07:
Hi again @laurensvalk Sorry for the slow response. It would be very much appreciated with a backport if that's still a possibility.

And can you provide me with the version number of the version that should fix the issue? :-)
thanks a lot

laurensvalk, 2024-03-07:
It's probably possible as a fairly small fix - please feel free to reach out to the email address listed above. Thanks!

BertLindeman, 2024-03-25:
From #1419 
Running Technichub at 3.5.0b1 firmware.

The small program that caused this rapid blinking (program A) ran on Hub-B:
```python

from pybricks.hubs import TechnicHub
from pybricks.tools import wait
from urandom import choice

transmitter = TechnicHub(broadcast_channel=1, observe_channels=[2, 3])

while True:
    transmitter.ble.broadcast([choice([True, False]), choice([True, False])])
    TriggeredDistributor, TableReadyForTipp = transmitter.ble.observe(3) or [0] * 2
    wait(100)
```

There was a primehub next to it running a similar program.
See [comment in #1419 ](https://github.com/pybricks/support/issues/1419#issuecomment-2017951133)

Also a technicHub (HubA) running the larger test program for #1419 

After running the smal 'program A' for 4,5 hours the TechnicHub-B got battery-low blinking.
Long press to stop the program resulted in rapid blink and batteries needed to be taking out.

The original larger test program still runs on (5 hours now).

Sorry if this became a bit long...

Bert

BertLindeman, 2024-03-26:
Additional symptom: 
The program that was saved on the hub no longer starts after the hub got stuck on rapid blinking.

My scenario:

1. beta.pybricks.com: connect the hub
2. start the program (via the pybricks start button)
3. stop the program (via the pybricks stop button)
4. disconnect the hub
5. start the program by pressing the hub button
6. wait, wait, wait no idea how long. This time i waited about 4 hours 
7. press the hub button - nothing happens
8. long press the hub button and get rapid bink
9. take batteries out and reset them
10. press the hub button - flashing (slow to indicate ready for connection)
11. try again press, but nothing happens.

No idea if the program got wiped, or something in the startup changed.
I gamble the program got invalid or wiped.

laurensvalk, 2024-03-26:
As explained in a separate thread, the program is saved during a successful shutdown. So if your hub crashes before that, your program is not saved. You can just do a normal shutdown at any time to properly save your program :slightly_smiling_face:

BertLindeman, 2024-03-26:
> As explained in a separate thread, the program is saved during a successful shutdown. So if your hub crashes before that, your program is not saved. You can just do a normal shutdown at any time to properly save your program 🙂

To be clear: with shutdown, you mean power-off of the hub. I took it for stooping the program in the standard way.
So I need to add a step to my test 😄 power-off the hub.
Sorry....

laurensvalk, 2024-03-26:
@BertLindeman - No worries, your testing is very much appreciated!

@SuneLindhard  / @sultrifork : Your code did not use Bluetooth, correct? Did the most recent firmware fix the issue for you?

sultrifork, 2024-04-03:
> @BertLindeman - No worries, your testing is very much appreciated!
> 
> @SuneLindhard / @sultrifork : Your user code did not use Bluetooth, right? Did the most recent firmware fix the issue for you?

Indeed it doesn't use Bluetooth. We haven't done anything really to disable it, if that's what you mean though

laurensvalk, 2024-04-03:
Right - and has the shutdown problem gone away after you upgraded the hubs in the setup?

sultrifork, 2024-04-04:
> Right - and has the shutdown problem gone away after you upgraded the hubs in the setup?

Unfortunately it has been decided to not upgrade the firmware for now, as it doesn't happen too often and the workaround is acceptable

laurensvalk, 2024-04-11:
I think we can close this issue as fixed. Thanks for reporting!

> Unfortunately it has been decided to not upgrade the firmware for now, as it doesn't happen too often and the workaround is acceptable

Happy to help with this. If anything, driving around and using sensors should have gotten a lot better since two versions ago :smile: 

In any case, I look forward to seeing this setup on my next visit!

closed 2024-04-11 by laurensvalk