pybricks/support #1498: [Feature] Add example for IR Seeker

laurensvalk, 2024-03-01:
> Hi I am going to use Pybricks blocks with my students for RoboCup Junior Australia Competition. In the soccer comp we will be using the IR sensor from [https://buildingblockrobotics.com/products/ir-seeker](https://buildingblockrobotics.com/products/ir-seeker?fbclid=IwAR1zLAVbMcAn4EIzkKoSE8NnqUNZYlI76nRiFkPSQ1zMWfYIt9c4YA0tagA)
>
> It has a sample to read in the ball but could you add the additional blocks to set up please.

_Originally posted by a Facebook user via [our page](https://www.facebook.com/Pybricks)_

BertLindeman, 2024-03-01:
> I think we can start by explaining how you can do it by using that sample in Python and importing it into a block program.

Is there already doc about the import into a block program?
Do you want everyone to know that trick?
I could make an example here from the IR-seeker example:

```python
from pybricks.iodevices import PUPDevice
from pybricks.parameters import Port

device = PUPDevice(Port.A)

while True:
    values = device.read(5)
    if len(values) > 0:
        print("Strength", values[0]) #return the signal strength
    if len(values) > 1:
        print("Direction", values[1]) #return the direction of the ball 0 - 12
```

No IR-seeker in my inventory, but I have a color sensor, and as documented it acts like a color sensor.

laurensvalk, 2024-03-01:
Thanks Bert. I received the sensor although the ball seems to be out of stock in most places. I found that it is able to produce values using an old radio remote, so I'll use that for basic testing.

> Do you want everyone to know that trick?

Sure! There's a few usability questions to sort out still though, so I haven't posted about it extensively.

Basically, importing functions works great, but right now the main block program may either be fully async or not, and the imported function has to match or it will not work as expected.

laurensvalk, 2024-03-01:
As I was writing that, I think a reasonable solution that doesn't require too much magic behind the scenes is #1499.

Beginning teams can just write synchronous code, but more advanced teams or third party manufacturers can use this make classes that work either way without having to think about it.