pybricks/support #1500: [Bug] drivebase with gyro enabled does turn as expected

DanieleBenedettelli, 2024-03-02:
**Describe the bug**
What is the problem?
We have a robot with treads and rubber pads that shakes a bit while turning,
We set up the Drivebase class with the correct dimensions.
We enable the gyro and ask the robot to spin by 360 degrees.
The robot spins by 340 degrees and stops. The gyro reads 340 degrees.
So the heading is estimated correctly, but the drivebase turn function has a bug.
It improves a bit by tweaking the axle_track dimension, but for some large values (e.g. over 200mm), the interpreter it gives an error
"Invalid argument".

**To reproduce**
I don't know if it's a matter of vibration during the movement (it works fine with hard wheels).
We even tried increasing the ang. speed and acc. in the IMU settings to no avail.

**Expected behavior**
The robot should spin 360 degrees.

**Screenshots**
I was coaching students in video call and I have no screenshots.

laurensvalk, 2024-03-03:
Thanks for reporting. Can you specify which dimensions and wheels were used?

DanieleBenedettelli, 2024-03-03:
Wheel diameter (tread height from rubber to rubber bump) was 47mm. Axle track was 135, then increased. The problem arises for values above 190 or 200. 
It disappears when increasing the wheel diameter.

laurensvalk, 2024-10-01:
I built the following masterpiece of a robot :smile:, but I can't seem to reproduce it. That is to say, I wanted to verify a small wheel with a wide base, but this does not appear to be problem.

https://github.com/user-attachments/assets/a70977d2-aa1a-4d3e-9f52-c79bd2a74750

laurensvalk, 2024-10-01:
Here is the worst drivebase in the world, with excessive wheel slip. Still comes out at 360:

https://github.com/user-attachments/assets/50e1cc0a-1f3b-4177-a2c0-e3570022c53a

BertLindeman, 2024-10-01:
Laurens, your latest movie above does have a bit of vibration. 

Daniele specifically mentions:

> I don't know if it's a matter of vibration during the movement (it works fine with hard wheels).

What kind of wheels do you use @DanieleBenedettelli ? 
A photo of the model or even the type of wheels might help.

Bert

DanieleBenedettelli, 2024-10-01:
The problem for me was, if I remember well, that I swapped the motors in
the drive base constructor, or that I configure the gyro axis in the wrong
way, w.r.t. my model.

closed 2024-10-01 by laurensvalk