pybricks/support #1502: [Feature] Implement Car class in C

laurensvalk, 2024-03-03:
**Is your feature request related to a problem? Please describe.**
The [version of the Car class](https://github.com/pybricks/pybricks-micropython/blob/a844b32fbc5c1b5084fa64ce09eb168b6c257ffa/bricks/_common/modules/_robotics_extra.py#L5) we ended up with only has three small synchronous methods. This should be easy and cheap to implement in C.

If we really saved 4KB on move hub, it is probably worth including this class. Both because it is useful and because it is a lot easier to have the corresponding blocks "just work" without exceptions.

Despite being a last minute change, this should be fairly low risk and will expedite the release of the docs and block code, so it is probably worth the time.

laurensvalk, 2024-03-05:
Done via https://github.com/pybricks/pybricks-micropython/commit/3281b1bf55d5679f59d7e1a3d9e8f8b5f18040e3

BertLindeman, 2024-03-07:
Ran your example from issue #1397 on the beta firmware on the movehub.
Great.
Was there any check on the motors intended? 
I made a typo in the car definition: used motorA both as steering motor and as part of the drive motors.
Silently the steer callibration is skipped and driving **is** done.
Took me some time to understand why the steering did not work ;-)

laurensvalk, 2024-03-07:
That's a great idea, thanks Bert!

laurensvalk, 2024-03-11:
@BertLindeman - released in stable firmware, accessible in latest Pybricks Beta. (Just some more test driving and then we'll call it stable!)

BertLindeman, 2024-03-11:
> @BertLindeman - released in stable firmware, accessible in latest Pybricks Beta. (Just some more test driving and then we'll call it stable!)

Did not see it as quickly as I wanted (also preparing "help, I got a smartphone" lesson..) 
Doing a test now.

BertLindeman, 2024-03-11:
Tested 
```
v3.4.0b3 (Pybricks Beta v2.4.0-beta.6)
```

Drive and steer motors should be uniqe.
Drive motors should be unique.

Tested via block coding.
All OK.