pybricks/support #1503: [Feature] Add integer division to math block

laurensvalk, 2024-03-03:
**Is your feature request related to a problem? Please describe.**
The Move Hub doesn't support division with `/` but it does support `//` for integer division.

**Describe the solution you'd like**
Add integer division as an option to the block dropdown.

[image]

**Describe alternatives you've considered**
Make `/` automatically work like `//` when floats disabled, but this is perhaps less explicit.

DanieleBenedettelli, 2024-07-17:
Hi.
When will this be implemented? 
Also in the equation block it would be needed:
[image]

Thanks

laurensvalk, 2024-07-18:
I've been focused on the new [Learn Guide](https://pybricks.com/learn/) lately, so realistically it might take a while before we can work on app feature updates.

andybellgbc, 2024-07-20:
It took me 2+ hours of head scratching and writing, rewriting, commenting out lines and Google searches to find out it wasn't me.

If you ignore my apparently poor decision to use the Boost hub!

At least make the blocks resolve correctly for the Boost Hub. While I see you have this an enhancement - for the Boost Hub this is a bug, especially because the error message does not provide meaningful help. If it at least had the line number.

Otherwise it's great 😃

laurensvalk, 2024-07-20:
Thanks for your note. I suppose we could make it automatic, but then it would only work if the user has selected the Move Hub block, which normally isn't a requirement.

Another nice (but more work) way would be to do it at compile time since we know which hub is connected at that time. By having mpy-cross catch it, we could even catch it for Python users.

Maybe there already is a mpy-cross flag for no floats? After all there is one for long ints I think.

dlech, 2024-07-20:
> Maybe there already is a mpy-cross flag for no floats?

No, the MPY bytecode would have to be parsed to look for float objects.

laurensvalk, 2024-07-20:
Just having a `//` option on the block is probably still the most sensible (and useful even on platforms that _do_ support `/`) but while my mind is at at... here's another.

---------------

Another trivial change would be to let `MP_BINARY_OP_TRUE_DIVIDE` fall through to `MP_BINARY_OP_FLOOR_DIVIDE` on platforms without floats (i.e. movehub only):

[image]

Then you'd get the following. Same for blocks and Python. This is relatively intuitive (decimal numbers are not supported, so division will truncate), but it isn't very Pythonic.

[image]

laurensvalk, 2025-02-25:
🎉 

[image]

laurensvalk, 2025-02-25:
This also adds descriptions the dropdown while we are at it. This is something we can do now that we're no longer using the standard blockly block. For comparison, old version:

[image]

BertLindeman, 2025-02-25:
Also make it easier to spot the difference between plus and divide. Nice improvement.

closed 2025-02-26 by laurensvalk