pybricks/support #1524: [Bug] XBOX controller button pressed translated wrong in python

DanieleBenedettelli, 2024-03-07:
**Describe the bug**
What is the problem?
[image]
The controller button reporter blocks are translated into:
`if "Y" in controller.buttons.pressed():
        lifter.run_target(200, 140, Stop.BRAKE)
        print('up')
    elif "A" in controller.buttons.pressed():
        lifter.run_target(200, 0, Stop.BRAKE)
        print('down')
    else:
        pass`

while the correct code should be:
`if Button.Y in controller.buttons.pressed():
        lifter.run_target(200, 140, Stop.BRAKE)
        print('up')
    elif Button.A in controller.buttons.pressed():
        lifter.run_target(200, 0, Stop.BRAKE)
        print('down')
    else:
        pass`

The translation is wrong for every button of the controller

BertLindeman, 2024-03-07:
Daniele,

I simulated this at:
```MicroPython for LEGO® Powered Up smart hubs.

v3.4.0b3 (Pybricks Beta v2.4.0-beta.6)

Copyright (c) 2020-2023 The Pybricks Authors
```

[image]

The python code generated:
```python
from pybricks.iodevices import XboxController
from pybricks.parameters import Button

# Set up all devices.
controller = XboxController()

# The main program starts here.
if Button.Y in controller.buttons.pressed():
    print('Y')
elif Button.A in controller.buttons.pressed():
    print('A')
else:
    pass

```

Are you using the stable code.pybricks.com?

Bert

laurensvalk, 2024-03-08:
Thanks for reporting @DanieleBenedettelli.

Coincidentally, this was fixed yesterday! Your version might not have reloaded just yet.

[image]

Thanks @BertLindeman for confirming that it was fixed.

closed 2024-03-08 by laurensvalk

DanieleBenedettelli, 2024-03-08:
Exactly that.
Once SHIFT+F5 reloaded, it worked.

D

laurensvalk, 2024-03-08:
Just opening the app again should automatically load it, so you don't even have to wait :smile: