pybricks/support #1531: [Feature] Allow descriptive text in function definition and caller

laurensvalk, 2024-03-08:
**Is your feature request related to a problem? Please describe.**

It could be nice if there was a way to label inputs, so the callers are more descriptive.

[image]

**Describe the solution you'd like**
- we could allow users to define the signature
- we could make them look like keyword arguments on the caller: `my function ( speed = ... , angle = ... )`

BertLindeman, 2024-03-08:
Probably a bit too far for now. 😉 

But how about a docstring to document the function?
Or even going further.. A docstring for the program.

Bert

laurensvalk, 2024-04-04:
See also https://github.com/orgs/pybricks/discussions/1582

laurensvalk, 2024-04-06:
See also https://github.com/pybricks/support/issues/1530 which we could do when we look into this.

We should work with labels instead of fields.

donmerendolo, 2024-12-06:
Hi,

Related to this (I think) are keyword arguments (at least I haven't found a way to use them).

I have been testing the blocks for my team (and I find them great for learning! 😃) and my idea was to import the functions we used last year for the movement of the robot. Most of them have a few positional arguments like `target_distance` or `target_angle` and a bunch of keyword arguments like `speed` or `wait`. Haven't found a way to use keyword arguments in blocks so they are inaccessible right now (which is not that important, we can convert them to positional for now).

Thank you! ☺️