pybricks/support #1532: [Bug] View resets to top right on open if first block has negative position

laurensvalk, 2024-03-08:
**Describe the bug**

Sometimes when opening a program, the view resets and pushes the program into the top left.

**To reproduce**

This also happens in upstream Blockly. It seems to happen only if the first block in the program has a negative X / Y position.

https://github.com/pybricks/support/assets/12326241/8d74edb5-37eb-4e07-9913-a64d1fe3a132

laurensvalk, 2024-03-08:
Related: https://github.com/pybricks/support/issues/1518

BertLindeman, 2024-03-10:
Seen the same on a block program where the first block has "x":0 and "Y":0 no other block has a negative position.

<details><summary>example program</summary>

```python
# pybricks blocks file:{"blocks":{"languageVersion":0,"blocks":[{"type":"blockGlobalSetup","id":"bjK,wS1MYO7aiYkFSwd{","x":0,"y":0,"deletable":false,"next":{"block":{"type":"variables_setup_any","id":"vXNa6W`)4cX-JAEa/k~y","fields":{"VAR":{"id":"grTpUun8%ue/cH-JVk?4"}},"inputs":{"VALUE0":{"shadow":{"type":"blockMathNumber","id":")G-TYEb.`J}yMNvy+yyf","fields":{"NUM":0}}}},"next":{"block":{"type":"variables_set_stopwatch","id":"/8PFO|u#?x[KEy^O|~VN","fields":{"VAR":{"id":"ieAQkvTm:WpmCC;WhY+7"}},"next":{"block":{"type":"variables_setup_imported_function","id":"i[323@J4M*/GJrib](Mx","fields":{"FROM_MODULE":{"id":"p(8R+Z(BvwNG[%O=k!-I"},"VAR":{"id":"_h[`0DIk@`O59N{a{,1K"}},"next":{"block":{"type":"variables_set_technic_hub","id":"~4MVinldJZE_2gMfR1Qs","extraState":{"optionLevel":0},"fields":{"VAR":{"id":"`#@a@?tUjM14U0z:PnZx"}}}}}}}}}}},{"type":"blockGlobalStart","id":"3tJe|AWl0baN(wH9a$@.","x":0,"y":576,"deletable":false,"next":{"block":{"type":"blockFlowForEach","id":"xFPfNGBH|:)K1@$W@Ztp","inputs":{"VAR":{"shadow":{"type":"variables_setup_iterable","id":"cnLU)ESy]LA5L970:WD}","fields":{"VAR":{"id":"~!)aMILdlG2zOjVl`REO","name":"item","type":"LoopVariable"}}}},"LIST":{"shadow":{"type":"blockListEmpty","id":"s(L/n?vun[J:[[Kd}@_g"},"block":{"type":"blockRange","id":"],Yklso)F^-44S1o^P]s","extraState":{"optionLevel":0},"inputs":{"RANGE_TO":{"shadow":{"type":"blockMathNumber","id":"tOJ9`QS[#fiIwz*M~l1P","fields":{"NUM":10}}}}}},"DO":{"block":{"type":"blockStopWatchDo","id":"76/NlZm`wbJ^th:x.!C^","extraState":{"optionLevel":0},"fields":{"METHOD":"STOPWATCH_RESET"},"inputs":{"VAR":{"shadow":{"type":"variables_get_stopwatch","id":"JPHmnL^4ZE0J#.Fv6SAj","fields":{"VAR":{"id":"ieAQkvTm:WpmCC;WhY+7","name":"watch","type":"StopWatch"}}}}},"next":{"block":{"type":"blockFunctionCallerStatement","id":"2Lt3XnWAGVpH8NgY:{9Z","extraState":{"optionLevel":2},"fields":{"ICON":"SUITCASE","VAR":{"id":"a6z/~FHla*.H$rok|)T5"}},"inputs":{"ARG0":{"shadow":{"type":"blockMathNumber","id":"fb_4)QD++F-(S~_,@.-:","fields":{"NUM":4}},"block":{"type":"blockMathArithmetic","id":"[(A]_%}h,y%M%!#~v{!/","fields":{"OP":"ADD"},"inputs":{"A":{"shadow":{"type":"blockMathNumber","id":"{#~;TkI/@_|j5)B8QfD,","fields":{"NUM":1}},"block":{"type":"variables_setup_iterableConstant","id":"]%]d(g)G`j{oq)nBwfQ`","fields":{"VAR":{"text":"item"}}}},"B":{"shadow":{"type":"blockMathNumber","id":"4r#Y5vCXx,N`zCtB{g,m","fields":{"NUM":4}}}}}},"ARG1":{"shadow":{"type":"blockMathNumber","id":"}Hjtsrxb[YaZ2Vso!_nH","fields":{"NUM":123456789}}}},"next":{"block":{"type":"blockIfElse","id":"N|Vy=oFq-~B2w{r[T32-","extraState":{"optionLevel":0},"inputs":{"IF0":{"shadow":{"type":"blockLogicTrue","id":"EU8_UJL2wh,p;8m{|VSx"},"block":{"type":"blockLogicCompare","id":"]Qa^n:lK1d`.pndMgG}a","fields":{"OP1":"LTE"},"inputs":{"A":{"shadow":{"type":"blockMathNumber","id":"6)O2M1]UbJi[qJHq-F`^","fields":{"NUM":3}},"block":{"type":"variables_setup_iterableConstant","id":"?*pK92QtGahox.Xo@,J.","fields":{"VAR":{"text":"item"}}}},"B":{"shadow":{"type":"blockMathNumber","id":"l3$2$uO/oKrza1*uC,|(","fields":{"NUM":0}}}}}},"DO0":{"block":{"type":"blockPrint","id":"?Plfb)PjoKm[O`2Gs/;u","extraState":{"optionLevel":3},"fields":{"SEP":"SEP_SPACE"},"inputs":{"TEXT0":{"shadow":{"type":"text","id":"`aLptf8rsVqA;f+Z9+Ce","fields":{"TEXT":"hub version"}}},"TEXT1":{"shadow":{"type":"text","id":"Sxs;*_(0rA9kJ4zDb-c}","fields":{"TEXT":"arg1:"}},"block":{"type":"blockreadOnlyGetter","id":"5%id5.4ZvYP~^Fxzb:z#","fields":{"VAR":{"id":"_h[`0DIk@`O59N{a{,1K"}}}},"TEXT2":{"shadow":{"type":"text","id":"[:;:I9@O*^`pl+Zq2^+d","fields":{"TEXT":"abc"}},"block":{"type":"variables_setup_iterableConstant","id":"y=b*,(p2wO)G%EPcxwk]","fields":{"VAR":{"text":"item"}}}},"TEXT3":{"shadow":{"type":"text","id":"}LYbocAM98YHN9,S5HS|","fields":{"TEXT":"abc"}},"block":{"type":"blockLogicCompare","id":"c0d*A)0g+}NXXMvgrwdg","fields":{"OP1":"LTE"},"inputs":{"A":{"shadow":{"type":"blockMathNumber","id":"uD?=3o~u*ny*}R25=*cB","fields":{"NUM":3}},"block":{"type":"variables_setup_iterableConstant","id":"MPkHpmz,HjWn1%Q%eo`P","fields":{"VAR":{"text":"item"}}}},"B":{"shadow":{"type":"blockMathNumber","id":"=-!?pa[QXpmBcC]#v%Zm","fields":{"NUM":0}}}}}}}}}}}}}}}}}}}},{"type":"variables_setup_function","id":".)|_p#q!*(N]K^v[=U/)","x":0,"y":275,"extraState":{"optionLevel":2},"fields":{"ICON":"SUITCASE","VAR":{"id":"a6z/~FHla*.H$rok|)T5"}},"inputs":{"ARG0":{"shadow":{"type":"variables_setup_function_argument","id":"Os$,f3zKMAkJkXxrrh%H","fields":{"VAR":{"id":"b7erH[gE|jzwZ?jU^;KO","name":"arg1","type":"FunctionArgument"}}}},"ARG1":{"shadow":{"type":"variables_setup_function_argument","id":"@|HrJ[/o_n|vH-3F`Ko+","fields":{"VAR":{"id":"mg$D3#y3?i.8rnQg)awU","name":"arg2","type":"FunctionArgument"}}}},"STACK":{"block":{"type":"blockPrint","id":"IG0SiC0^3?;)pKQ!yw(J","extraState":{"optionLevel":4},"fields":{"SEP":"SEP_SPACE"},"inputs":{"TEXT0":{"shadow":{"type":"text","id":"PZ.^E|Vo,C%3fh?ml[}B","fields":{"TEXT":"in my func_1"}}},"TEXT1":{"shadow":{"type":"text","id":"m:T7pTZl$lxJkhuhZE4x","fields":{"TEXT":"arg1:"}}},"TEXT2":{"shadow":{"type":"text","id":"Q]}1~W!Ps+o~ECL/dx?D","fields":{"TEXT":"abc"}},"block":{"type":"variables_setup_function_argumentConstant","id":"E%%8{|bAk,n0hxh;Mh#=","fields":{"VAR":{"text":"arg1"}}}},"TEXT3":{"shadow":{"type":"text","id":" [... cut ...]
    func_1_(item + 4, 123456789)
    if item <= 0:
        print('hub version', version, item, item <= 0)

```

</details>

Not a real problem, I think.

A scenario I did: 

- in the block program press the (how is it called?) icon with 3 thick stripes, so the program goes to the center of the screen

- switch to another tab in the app or browser.

- switch back to the block program and see the program shifted to the left top in the window and the size may be adapted. (I saw it getting zoomed larger)