pybricks/support #1542: [Bug] Block coding font is different across platforms

laurensvalk, 2024-03-12:
> [image]

_Originally posted by @BertLindeman in https://github.com/pybricks/support/issues/1541#issuecomment-1991231447_

laurensvalk, 2024-03-12:
It looks like this for me:

[image]

BertLindeman, 2024-03-12:
> It looks like this for me:

Hard to say I am not that color-proof. So I took another screenshot and showed the result next to pybricks
and took another windows screenshot, maybe this helps:
[image]

Pybricks on the right side.
Looking at the green I think "my" green is a bit faded. (in Dutch: wat flets)

BertLindeman, 2024-03-12:
**To me**! the code blocks look the same on win11 and on an android tablet.

Also took a code block photo on android of the same program:

[image]

On android the background is transparant ;-)