pybricks/support #1543: # Manually installing USB driver in Windows

paulusBos, 2024-03-13:
# Manually installing USB driver in Windows

As describe above, if you see a warning sign in the *Device Manager* next to `Lego Group LEGO Technic Large Hub in DFU Mode` like this, then you need to manually install the USB driver for this device.

[image]

1. Double-click the device to open the *Properties* page.

   [image]

2. Go to the *Drivers* tab and click the *Update Driver* button.

   [image]

3. Select *Browse my computer for drivers*.

   [image]

4. Select *Let me pick from a list of available drivers on my computer*.

   [image]

5. Select *Universal Serial Bus devices* and then click *Next*.

   [image]

6. Select *WinUsb Device* under *Manufacturer* and again under *Model* then click *Next*.

   [image]

7. Accept the warning by clicking *Yes* (we have verified that the driver is compatible).

   [image]

8. The driver should successfully install and you can click *Close*.

   [image]

_Originally posted by @dlech in https://github.com/pybricks/support/discussions/688#discussioncomment-3201466_

paulusBos, 2024-03-13:
Can't find WinUSB
[image]
 device

laurensvalk, 2024-03-13:
@paulusBos - thanks for asking!

I can’t seem to find your post with the question anymore. Were you able to find it?

What was the solution? Others might have the same question. Thanks!

paulusBos, 2024-03-13:
[image]

was looking for the correct WinUsb device 
but startted wrong (Windows in Duthc language)

laurensvalk, 2026-10-08:
In the latest version beta.pybricks.com, installing the driver is no longer required so we can close this. Thanks!

closed 2026-10-08 by laurensvalk