pybricks/support #1550: [Feature] Add block or method to reset drivebase odometry

laurensvalk, 2024-03-19:
**Is your feature request related to a problem? Please describe.**
The `DriveBase.reset()` method is not currently available as a block

**Additional context**
Requested by @MonongahelaCryptidCooperative

laurensvalk, 2024-03-19:
Proposed block:

[image]

MonongahelaCryptidCooperative, 2024-03-19:
Exactly what is needed. With this and the ability to change the shit down
button we should be able to do our FLL block starter pack code without
having to import anything. It will make it much less intimidating for new
teams.

closed 2024-03-21 by laurensvalk

laurensvalk, 2024-03-21:
@MonongahelaCryptidCooperative - this is now available in the latest beta version.

[image]