pybricks/support #1552: [Feature] Prepopulate toolbox with color equality test

laurensvalk, 2024-03-19:
**Is your feature request related to a problem? Please describe.**
We currently have a read color block and a color literal instead of a boolean "is this color block".

This provides for a lot of flexibility, but it can be tricky for new users. Instead of making a new block, we can make the union of that commonly used combination of blocks directly on the toolbox.

laurensvalk, 2024-03-21:
So we could change this:

[image]

To:

[image]

After you drag this out, you can still get each of the blocks separately.

laurensvalk, 2024-03-21:
@ggramlich - would you like to see the separate blocks on there, as well?

ggramlich, 2024-03-21:
@laurensvalk 
Keeping the (sensor color/hsv) block in the toolbox would be nice.
For the (red...) color block - I don't think that you actually need it as a separate block, if you can drag it out of the comparison block.

BertLindeman, 2024-03-21:
Laurens, this example above showed me how to do the color test.

If I would only want to print the color, I would need to pull  a color-comparison block, pluck it apart and then place the color-block into a print.
Not really beginner-friendly, I think.

I would prefer the separate color block as well in the input box.

laurensvalk, 2024-03-21:
Like this?

[image]

ggramlich, 2024-03-21:
@laurensvalk That would be fine for me.
I am not sure, if @BertLindeman also wanted the color block itself to be available separately.

BertLindeman, 2024-03-21:
> @laurensvalk That would be fine for me. I am not sure, if @BertLindeman also wanted the color block itself to be available separately.

I would the color block for beginners sake separately, but I am not fanatic about it.
Children learn so quick.

laurensvalk, 2024-03-21:
Thanks both of you for your insights!

laurensvalk, 2024-03-23:
Probably also need the color block separately as well, see https://github.com/orgs/pybricks/discussions/1563#discussioncomment-8887567

laurensvalk, 2024-03-29:
Other block brought back too now.

laurensvalk, 2024-05-08:
When you drag the populated block onto the canvas, it doesn't run the callback that automatically picks an available sensor.